vlachoudis/bCNC #309: Probing with ID 33 error

danst0, 2016-04-30:
Hi,

I am struggling with an GRBL error. However I think it might be related to bCNC. I try to probe my new tool in the beginning or after tool change.
The commands / errors are as follows:

> g53g0z0.0
> g91G38.2f20.0z-0.0
> g4p1
> ok
> ok
> ok
> ok
> error: Invalid gcode ID: 33
> ALARM: Hard limit

The detailed error is shown in the screenshot:
[image]

This error is thrown no matter if I am already at the X/Y-Probing position or if I am away.

Daniel

vlachoudis, 2016-05-02:
From your output it seems you are starting the probe from Z=0.0 until Z=0.0

closed 2016-05-03 by vlachoudis

danst0, 2016-05-03:
I am still unsure what I am doing wrong. Picture 1 is directly after the homing cycle. 
[image]

After that I press the calibrate button. Then the ID 33 error shows up immediately, however the X/Y move to the Probing coordinates is still performed, then the machine stops (picture 2).

[image]

vlachoudis, 2016-05-03:
You have to set a value in the "Distance:" field. This is the probing distance to travel from "Probe:MZ" to "Probe:MZ - Distance". Since it is 0.0 the probing fails immediately. I believe is a good point that I should protect it, or at least issue a message to the user.

danst0, 2016-05-03:
Thanks for the quick feedback! Now it works!

Daniel

vlachoudis, 2016-05-03:
Great. I've added also a warning message if it is not set :)