vlachoudis/bCNC #310: Feature Questionaire

vlachoudis, 2016-05-03:
bCNC is approaching to V1.0, containing all the features I wanted to have in the first place and much more. So my question to you is what extra features, improvements you would like to see in bCNC or even things that exist to and to be removed.
I should have a made a doodle poll for voting but for the moment I will leave it here as an issue. Unless if someones knows a way to do it in github.
Below is some items that come up to my mind, please feel free to add new ones, vote what you want

1. Hardware pendant support
1. Better Camera support
1. Camera calibration for better camera alignment
1. Simplifications on the Interface
1. Tablet support
1. More CAM features (like differences of paths, tabs to stick on path....)
1. Carve
1. Line alignment
1. Importing SVG files
1. Speed improvements (making critical functions in C)
1. ...

Effer, 2016-05-03:
Adding some more to the list, I'll vote after further thought

11. Optimize by drawing only bounding box for groups (when CPU is stressed)
12. Shape generator (Rectangle, Circle, Stars) from toolboox
13. Rotate, Scale, Deform from toolboox
14. Snap to grid
15. Nesting, optimize group locations to reduce waste of material
16. Split gocde, for small machine
17. Windows executable (py2exe)
18. Trace image (vectorizer)
19. Slice 3D STL model 
20. Basic 3D Cam

Hope you don't plan to stop after 1.0 anyway

flaplante, 2016-05-03:
>  2. Better Camera support
>  3. Camera calibration for better camera alignment
>  4. Simplifications on the Interface
>  5. Tablet support
>  6. More CAM features (like differences of paths, tabs to stick on
>  7. Carve
>  8. Line alignment
>  9. Importing SVG files
> 10. Speed improvements (making critical functions in C)

Speaking as a retired developer, I say add no more and put this to bed 
as 1.0.

Otherwise the feature crawl will go on forever, each adding its own set 
of new bugs.

You have done a fabulous job with bCNC and I would recommend it if it 
never added a new feature.

Fred LaPlante, WA1DLZ

atrueresistance, 2016-05-04:
HAHA @flaplante  Yeah well said! bCNC is awesome! I haven't seen really any bugs aside from the rendering large files not showing whole rendering, which I believe got fixed... and the serial thing, which effects a very small number of folks. 

Here is my votes 
1) Release, and take a week off. You devs deserve it!
2) speed improvements for rendering. 
3) Copy object with dimension capability.  You can already duplicate an object with bcnc, but setting spacing is kind of difficult if you want you know say 1.257" over for instance. Something like the dimension tool in Fusion 360 would be awesome.
4) Handle serial disconnects more gracefully. (I have one of those SO3 units) 
5) I'm guessing since @Effer mentioned it there is no way to scale gcode currently, this would be awesome. Take into account Z as well.
6) Web interface, more modern, allow list of previous files to be loaded without immediate access to files. I tend to access from my phone most of the time.

vlachoudis, 2016-05-04:
@Effer wants to bring bCNC to another dimension :)
@flaplante I want to stabilize it, and only introduce some minor things before naming it v1.0 
@atrueresistance once you copy an object. Go to the command line and type "move dx dy" where dx dy is the distance to move. More modern web interface is out of my knowledge :)

I forgot to add also automatic configuration from grbl parameters or vice versa

ErikShorb, 2016-05-04:
bCNC, first off, great tool.  Many thanks for your efforts.

My request would be for a more workflow oriented GUI/UX.  There's a fair
amount of bouncing between menus/sub-menues during normal use, and some
items seem to be less obviously grouped: i.e., Unlock, reset, home on
"Control" page, serial connection open/close on "File" page.

Suggest:

   * Control page with
      * Serial connect (on bottom),
      * "home" (rather dangerous button, should have "are you sure" dialog)
along with unclock and reset as part of Serial connect section
      *  Probe, and autolevel  as part of Control page
      *  File open replacing Home/unlock/reset buttons at top of Control
section.

The idea being you could reasonably fit all of the normal activities for
PCB work on a single menu/tab page.

If reorganization would cause grief, perhaps simply adding a PCB menu/tab
organized roughly as mentioned above for those focused on such use.

vlachoudis, 2016-05-05:
@ErikShorb the idea behind splitting it into multiple tabs was to group all operations and to minimize the space needed per page/tab. However you can re-configure bCNC to your needs. You all need to do is to change in the $(HOME)/.bCNC which groups and frames to appear in which page. It can be done be changing (or setting in the [bCNC] section the variables control.ribbon, control.page, etc. Look the bCNC.ini for the names.

But one idea will be to let the user create custom tabs.

Effer, 2016-05-05:
And add a user friendly way to customize the GUI.

PeterStraub, 2016-05-05:
Dear Vasilis Vlachoudis

Thanks for that fantastic work. I really like bCNC and it has amazing features. 

I just started to set up bCNC on a Rasperry Pi3. 
Start installation went easy. 

Of what i have seen and used up to now, it's just awesome. 
For beginners, maybe some help or manual for one or the other function or installation would be fine. 
Like the definition of function-keys. 

Ok, the original 7 inch display is a bit to small. 
- so I am looking forward to your interface adoptions. But i suppose, a bigger display will be the solution. 

Xbox-controller works fine, after some isues installing. 

My USB-camera for x/y referencing, still is on strike, cant find a picture in bCNC. Might be, because i got no Linux knowledge, but i'm improving. 
- the better camera support sounds helpful. 

If you can include, what you mentioned, that would be absolutely cool. 

Thank you very much

Best regards from Switzerland
Peter

tmie, 2016-05-11:
First: big thanks for your work !

I have one thing I would like in addition to only selecting the origin of a workpiece with a camera.
I want select multiple points of the target with the camera and the software rotates the milling paths to match the orientation of the part on the machine.
With this feature you would no longer need to get a perfect alignment of an item if you forgot for example one pocket in the initial milling process and you need to do some rework. You would just place the piece on the table with approximate orientation, tell the software where it is located and what the angle is by selecting some identification points.
I know, this is a big request, but maybe, sometimes in the future .....

Best regards from Germany

Tom

vlachoudis, 2016-05-11:
@tmie yes it is a big request, but it already exists in bCNC :)
Check the wiki https://github.com/vlachoudis/bCNC/wiki/Probe-Camera-Alignment at the bottom of the page.

Effer, 2016-05-11:
I throw in the pot: backlash compensation? I have a belt system, but maybe someone might benefit.

tmie, 2016-05-11:
>  but it already exists in bCNC :)

Uuups ... I was searching for it, but did not see it. 
Sorry for that and thumbs up !!!!

Greetings 
Tom

vlachoudis, 2016-05-12:
Don't worry @tmie bCNC is lacking on documentation. As https://xkcd.com/1343/ explains we need to find a clever balance

tmie, 2016-05-12:
Because my last wish was too easy ;-) I have another idea for an improvement I would like, but don't know if it is only my hobbyist view or if it would be also interesting for the other users:
The jogging in the way its implemented in GRBL is jumping in fixed steps. 
Instead of this discrete movements I would prefer a more analog way of control, preferrably with the XBOX controller (like implemented in the ESTLCAM software: https://www.youtube.com/watch?v=_y4g8zBoOV0 at 6:15min).
I know that a perfect implementation of this type of control depends on an implementation on the Arduino side, but maybe you could mimic this behaviour by a trick:
On the timebase of, for example, 200msec read the current position of the machine from the GRBL controller. After this read the percentage of the throttle axis of the joystick and check what axis is selected to move. Now calculate the moving speed of the machine and send it to the GRBL by the "F" command. Next, calculate how far the machine should move in the next 200msec and send the new target position to the GRBL controller. And so on and on....
With this behaviour you could move fast if the distance is long and release the throtte the nearer you get to the destination position in the intuitive way you are moving with your car.
I don't know if this is realistic or possible, but it would eliminate the need to check what stepsize is currently selected before moving and possibly smashing the tool into the table :-)

And I'm pretty sure I did not miss this feature while looking in the wiki ;-)

Best regards
Tom

tklus, 2016-05-25:
Just got this setup on my Raspberry pi 2 and it seems to be working great! Thanks for your hard work. It is going to be nice to have my CNC as a stand alone machine.

I would like to see a few changes to the gui. Would it be possible to have the "digital read out" a larger font size. when jogging and working from a distance it is hard to see location on the screen. and also, on the jog buttons, instead of arrows just have the buttons labeled with X+ X-, Y+ Y- and Z+ and Z-. something like the GRBL Panel controller software. 

It would also be nice to have a GUI for changing the GRBL settings.

Thanks again!
Tim

vlachoudis, 2016-05-25:
@tklus the fonts of the DRO  you can modify on the "Tools -> Config -> Fonts" if you have a digital font on your machine you can select it there, as well the font size. For the moment it doesn't have a gui to setup so you have to type in the value is "Family, Size, Bold"

tklus, 2016-05-26:
Awesome! Thanks

vlachoudis, 2016-06-09:
From #348 request a keyword to execute python commands only during run maybe something like "onrun"

Effer, 2016-06-09:
@vlachoudis for #348 what about a special symbol, like an *@* instead of *%*?

vlachoudis, 2016-06-09:
@Effer I don't like using many special symbols. Or maybe something that starts with % like
%run: 
 Unless if we populate a variable "state" or "run" holding a boolean value if the process is running or not, and you can use in the code 
%if run: subprocess.call("beep.exe")
the only drawback in this case it will be compiled and executed but doomed to fail in the editor.
Or we could intercept as a prefix "%if run:" and ignore the line.

Effer, 2016-06-09:
@vlachoudis  I'd like something that can be executed only when grbl really reach a specific line. A new symbol will encapsulate all the conditions (%wait, %if run: ,etc)  in a fast way and I hope will be more recognizable. The editor will skip the line.
I see a lot of applications for this feature: external I/O handling but even gcode rewind for repetitive task or conditional jump based on external conditions.

vlachoudis, 2016-06-10:
@Effer symbol makes life a bit easier when you know it, but once you forget it it is complicated. Explicit text is easier to read and understand.
The %wait is a "dirty hack" waiting for @chamnit to provide in v1.0 a nice way from the grbl side.

Effer, 2016-06-13:
Another useful option could be **spiral cut**: constantly move z down while cutting the material, instead of enter at fixed zstep.
A **feed & step calculator** would fit nice too. Now we need to change the depth increment from the material instead from the end mill option. (Thumbs rule, zstep = Tool diameter /2 )

vlachoudis, 2016-08-03:
@ThierryM you mean to have a simulate feature of the gcode?

ThierryM, 2016-08-03:
Yes but like a "movie" (an animation) to see automatically the route of the bit from the beginning to the end (like playing the route).
Regards

onekk, 2016-08-14:
Hi to all, I'm not dead only busy with life and so on (and working on the CNC too)

my opinion, based on my recent use of bCNC:

1) Hardware pendant support (how? I'm using a SBC card and I have plenty of GPIO pin that i could use, my card is an Olimex Olinuxino-A20-micro, running Armbian (previously the stock Olimex debian based distro)
Another option that could be feasible, it think is a gamepad as there are many inespensive ones with USB connection under 10 or 15 Euros.
)
2) Simplifications on the Interface
3) Tablet support (if you intended as an app for Android to communicate with bCNC, and this option could replace or pair with the Hardware Pendant) 
4) Speed improvements (making critical functions in C) maybe for the calculus function.

But one thing I think is the most wanted, some tutorial on using the CAM functions, I haven't figured out how to use them, I'm used to use CamBam to make the GCode and then send the NGC file to bCNC via a usb key or recently through ftp to the CNC controller.

Best Regards and many Thanks for the good works.

Carlo D.

tuxintrouble, 2016-10-24:
First of all, thanks for bCNC – to me it is currently the best GRBL GUI out there.

I would love to see a tiny improvement on the jogging function as it is currently implemented:

Currently, if the increase on the current position is outside the limits of the workspace, GRBL responds with a soft limit alarm or even worse, if soft limits are not enabled, the machine runs into the end switches. It would really be helpful, if bCNC would check if the increased position hits the limit and if so, just do nothing, instead of sending the G0 command to GRBL. This way, the user would not have to go through the kill alarm / reset cycle, but could just decrease the jogging distance and try again. This would be helpful on the GUI but even more helpful for users who use a physical pendant, such as a gaming controller. I am also looking forward to any kind of interface that is about to come. I really like the socket interface idea.
Best regards and many thanks, Sebastian

chamnit, 2016-10-24:
@tuxintrouble : The upcoming Grbl v1.1 release has a new jogging command that will just ignore jog commands that exceed the soft limit volume, rather than alarm out. It should make controlling Grbl with a pendant much easier to implement. 

As a side note, I quickly put together a Wii joystick to stream jog commands to it. Worked really really well.

vlachoudis, 2016-10-25:
@tuxintrouble @chamnit this is my next goal to use the new jogging capabilities of grbl V1.1 as well replace the feed override with the new mechanism.

einencool, 2018-04-26:
Hello, I would like to ask, if it would be possible to add the feature to control the feed rate with a „pots“? (Is this really the word? In German it is „Potentiometer“)

With the Software „Estlcam“ there is a feature on the Arduino Mega on the Pin ADC 1 and 2 to control the feed rate and the spindle speed. Because I love your Software, it would be great if you could add this features. 
In „Estlcam“ there is altough the feature to jog the Spindle to a desired place with a „hand wheel pulse encoder“ and the Arduino Mega. 
This would be altough a great feature for future Versions of your wonderful program.

Greets Chris

Harvie, 2018-06-12:
@einencool this would have to be added directly on to arduino (grbl code, not bcnc). and that's probably not going to happen as grbl is not willing to add new features, because they've runned out of program memory on arduino.

aggiedan977, 2019-01-25:
I have found the documentation to be really good - THANK YOU! 

Would someone be able to direct me to a thread regarding setting up jogging with an external controller (xbox, etc..)?

luckypickle1, 2024-06-29:
I'm currently working with bCNC on a project for my needs, which basically uses a camera to recognize objects, in this case end mills for PCBs for an ATC (Automatic Tool Change) system.

MARIOBASZ, 2024-06-29:
It can be interesting to quickly find the origin of the workpiece. I am also with my ATC project. In principle I don't see the need to use a camera for this.