vlachoudis/bCNC #313: when in inch mode, move gantry still issues mm

wolfmanjm, 2016-05-06:
When the controller is in inch mode (G20) using the move gantry or measure it still uses mm.
This is particularly bad when you move the gantry to a spot on the work and it moves 25.4 times further than it should.
I have set inch mode in both the controller, and in the inch setting on the control page.

vlachoudis, 2016-05-13:
@wolfmanjm I forgot your comment :)
The "gantry" move command uses what is defined as units in the settings and they should match with the controller. Periodically is peeking the status of the controller. I could use this information to correct the move. However nothing will guarantee that the information it has, is up to date before the move.

wolfmanjm, 2016-05-13:
The controller is in inch mode as confirmed with $G, and the settings says inch mode, but it seems it still sends the move in mm. At least it does not move anywhere near where the point I clicked on. I'll try to get a console dump to prove it :)

vlachoudis, 2016-06-09:
One solution would be that the "gantry" command to prefix the motion command with the right units, with the risk that the user has set a different one.