vlachoudis/bCNC #314: Autolevel inaccuracy

InvCat, 2016-05-08:
I found an inaccuracy in autolovel. When I make a simple probe to get the zero height and then make an autolevel then it always report 0.018 - 0.024mm higher then zero.
I set the feed to 1, and i see it wrotes the height 0.006mm in DRO but it uses higher height in the canvas et the same X-Y position.
The simple probe repetition accuracy +-0.003mm

vlachoudis, 2016-05-08:
Can you check what is reported in the Terminal as probe height.

InvCat, 2016-05-09:
I made some pictures:
https://drive.google.com/open?id=0B2XgNQJweEN8ZWJiVV9wSkN6SFk

vlachoudis, 2016-05-12:
@InvCat what intrigues me is your second image. After the manual probe your machine stopped at mz:-0.781, while your machine stopped at mposz:-0.803, 22um lower. So by clicking "Z=0" on the next image of course you have always this shift of higher.
Maybe @chamnit can enlightens us. Is the machine after touching the probe goes to a decelerated stop?

chamnit, 2016-05-12:
@vlachoudis : Yes. The probing cycle does a controlled deceleration immediately after detecting the probe switching states. This is to ensure the Grbl and the CNC machine remains in position sync. There isn't a way around this, and the travel after the probe trigger is dependent on both the probe feed rate and the machine acceleration.

vlachoudis, 2016-05-12:
@chamnit of course otherwise it will imply infinite deceleration and will lose steps.
So the best approach is after probing before setting to Z=0 one should return to the probed location, by executing something like that in the command line.
```G53 G0 Z[prbz]```

chamnit, 2016-05-12:
@vlachoudis : If [prbz] is the starting point of the probing cycle, then yes! :) And just to be clear, the probe cycle records and reports the position when the probe is triggered, and not the stopping point.

vlachoudis, 2016-05-12:
[prbz] is the z coordinate of the reported [PRB:prbx,prby,prbz:1]
Normally this behavior doesn't change the autolevel algorithm, because I am using the grbl returned probed location. However when I am milling PCB, in the final step I was using a probe command to detect zero location and I was zeroing my WCS on where the machine stopped. Which will lead to a zero location inside the material (by a few 10um depending on the z-acceleration).

Maybe it can be a good suggestion for a user defined button to move back to the  probe location.

vlachoudis, 2016-05-13:
I am closing the issue. I've included the G53 G0 ... command inside the default "User" commands, and I am wondering if it is better to add it as separate Button inside the "Probe" box to make it more clear.
For the moment I close the issue

closed 2016-05-13 by vlachoudis

Effer, 2016-05-13:
IMO a new button inside the probe box will fit nicely.

InvCat, 2016-05-16:
Thank you for the solution.