vlachoudis/bCNC #315: bCNC 0.9.3 locks up sending gcode, and also locks up host computer.

spanner888, 2016-05-09:
When run is clicked to sending gcode, bCNC and host computer locks up.
- Occurs without spindle running.
- Occurs on two different computers, both fresh installs of lubuntu.
- Is repeatable and occurs every time.
- locks up = mouse cursor still moves but clicks do not work, keyboard no longer works, can't even see caps lock light change, can't ctl alt F1 etc
- locks up seems almost instant after pressing run button.

- Job does run to end - ie CNC completes all movements. Note jobs/gcode where very short.
- Dry runs without steppers enabled OK, ie bCNC does not lockup.
- Using UniversalGcodeSender with same hardware and gcode works fine.

Seems obvious this is not a widespread issue. Is there some additional debug/logs that can be collected?

 
**Configuration:**
Program     : bCNC
Version     : 0.9.3
Last Change : 12 Apr 2016
Platform    : linux2
Python      : 2.7.9 (default, Apr  2 2015, 15:34:55) 
[GCC 4.9.2]
TkVersion   : 8.6
TclVersion  : 8.6

Attached is one of test files used.
[grbl_example.gcode.zip](https://github.com/vlachoudis/bCNC/files/254814/grbl_example.gcode.zip)

Effer, 2016-05-11:
Hi @spanner888 ,
I can't test till the end of week, so in the meanwhile I've a couple of question:
- How long it takes to comple the job?
- Does the lock persist during all the job?

What really sound strange to me, is that the problem occurs only when stepper are enabled.

vlachoudis, 2016-05-11:
Is it related to the Intel card video problem #250?

spanner888, 2016-05-12:
First responses to your queries, then a lot of updated info:-
Example code uploaded takes 90 seconds.
Lock up persists during AND LONG after. Note cannot alt tab etc, keyboard lock light does not toggle, Linux console keys don't work (ie Ctl Alt F1-7) etc.
CNC will only run short time then stop, eg 2 circle cuts of a 15mm bowl. Presumably this is just gcode in grbl buffer.

Only when stepper enabled - hmm either that was poor testing/observation by me, or another change in behaviour. Now get crash without steppers or spindle running.
Note crash is different behaviour to lockup originaly descrbed. See below.

Not directly related to Intel card video problem #250, but maybe related via xorg, again more info below.

Updated testing information:-

    Problem has shifted from lockup to an "instant logout", which is probably an xorg crash - sometimes get a crash report for xorg appear on logging in again.
    This even occured on computer that is not internet connected, so no idea why the behaviour changed.

    Problem occurs on non-Intel video cards - have reproduced on three different nVidia cards, in two computers.

    Same computer running Lubuntu 15.10 or Windows 7 is OK, but running Lubuntu 15.04 crashes.
    The newer Lubuntu info is:-

    Linux 4.2.0-16-generic (x86_64)
    Distribution Ubuntu 15.10
    Desktop Environment LXDE (Lubuntu)
    ...
    Loaded Modules
    nouveau nVidia Riva/TNT/GeForce/Quadro/Tesla
    video ACPI Video Driver

    The older Lubuntu info is:-

    Linux 3.19.0-15-generic (i686)
    Compiled #15-Ubuntu SMP Thu Apr 16 23:32:01 UTC 2015
    Distribution Ubuntu 15.04
    Desktop Environment LXDE (Lubuntu)
    ...
    Loaded Modules
    nouveau nVidia Riva/TNT/GeForce/Quadro/Tesla
    video ACPI Video Driver

    Problem DID NOT occur on a old Lennovo model 9088ABM with Video card Intel Q33/Q35/G running OpenSuse 4.0.5.3, so it is not an issue with "every" Intel video card.

    Problem DID NOT occur on exact same hardware & software configuration, except running bCNC 0.5, instead of bCNC 0.9.3 - when problem DOES occur.
    Verified this on two of the above computers.

Why so much Lubuntu? Well the two computers that have been connected to CNC are really old - I got as extra unwanted items in pallet load of IT gear from an auction.
So using Lubuntu as it is supposed to work on older & low resource systems.

Novena desktop (see https://www.crowdsupply.com/sutajio-kosagi/novena/) running Debian 8 is OK.

Attached is xorg crash information from the Lubuntu/Windows computer: 
[2016-05-12-034825_1513x1018_scrot.png.zip](https://github.com/vlachoudis/bCNC/files/260846/2016-05-12-034825_1513x1018_scrot.png.zip)

Can provide more info, but probably best if someone can push me in correct direction, else I will flounder around....

Thanks for looking at this and especially for bCNC.

Effer, 2016-05-12:
I'm not a Linux user but when I tried Lubuntu I found it a nice choice. It's a pity there are some problem. 
Good you find it worked with 0.5 version, this will help a lot.

ChunkyBits, 2016-05-17:
A big thank you to the developer team for this software, it is just what I need. Far better than the Universal GCode Downloader ( in general). II thought I would let you know - I have the same problem.

The interface locks up after hitting RUN. There no way out of it. Although it continues to send the GCode. I have to pull the USB cable to stop the run, Then power-off the laptop.

The laptop is an old IBM thinkpad model 1834 ( not the year it was made ), running Xubuntu 14.04  ( Ubuntu + xfce ).

Also, if that doesn't happen, I get logged out instead. The job halts.

I tried it with no power to the spindle and in grbl check mode, the same results.

I have abandoned the laptop. Now I am using a Dell mini tower with exactly the same software set-up. I don't get get this problem.

I would be able to supply details if it would help.

vlachoudis, 2016-05-18:
I am using xubuntu 16.04, but the development started with xubuntu 14.04 and after. I never faced such a problem. Normally a program running from the user should not lock the computer. I can see two options:
1. Either it enters an infinite loop and runs out of memory. Do you use autolevel?
2. xorg is crashing for some unknown reason.
Seeing your bug report it seems it is the second. In this case it should be reported to xorg.

ChunkyBits, 2016-05-18:
Hello Vasilis,

1. No I wasn't using auto level.

2. Yes I agree, I don't think it's anything to do with bCNC.

I added my comment because it's behaviour is very similar to the original
bug report I thought you might like to have known about it. As I said I've
abandoned the laptop. I am using a PC so for now the problems are solved.

Thank you for your efforts, I like bCNC a lot.

I have a question. What version of Python is bCNC written with. I found out
this morning that although the default version in Ubuntu is Python3 since
ver14.04. In fact if you run python from the command line as  bCNC.sh  does
e.g.

steve@laptop:~$ python
Python 2.7.6 (default, Jun 22 2015, 17:58:13)
[GCC 4.8.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>>

So I am using Python ver 2.7.6. The symbolic link python in
/user/bin/python points to python ver 2.7

steve@laptop:~$ ls -al /usr/bin/python*
lrwxrwxrwx 1 root root       9 Feb  2  2015 /usr/bin/python -> python2.7
 ###### <- This one
lrwxrwxrwx 1 root root       9 Feb  2  2015 /usr/bin/python2 -> python2.7
-rwxr-xr-x 1 root root 3345416 Jun 22  2015 /usr/bin/python2.7
lrwxrwxrwx 1 root root      33 Jun 22  2015 /usr/bin/python2.7-config ->
x86_64-linux-gnu-python2.7-config
lrwxrwxrwx 1 root root      16 Dec 21  2013 /usr/bin/python2-config ->
python2.7-config
lrwxrwxrwx 1 root root       9 Feb  2  2015 /usr/bin/python3 -> python3.4
-rwxr-xr-x 1 root root 4061272 Apr 11  2014 /usr/bin/python3.4
-rwxr-xr-x 1 root root 4061272 Apr 11  2014 /usr/bin/python3.4m
lrwxrwxrwx 1 root root      10 Feb  2  2015 /usr/bin/python3m -> python3.4m
lrwxrwxrwx 1 root root      16 Dec 21  2013 /usr/bin/python-config ->
python2.7-config

Although I can change the symbolic link  e.g.
steve@laptop:~$ cd /usr/bin
steve@laptop:/usr/bin$ sudo ln -f -s ./python3 ./python

After having to install Tkinter3, and running bCNC your code has syntax
errors.

So I am puzzled. Should it be Python ver2.7.6 or ver3.4

Anyway, thanks again for a wonderful piece of software.

Steve H

vlachoudis, 2016-05-18:
@ChunkyBits bCNC is written in python v2, slowly we are trying to move to python v3

bhgv, 2016-05-23:
hi, i have same touble. Lubuntu, armhf, mali-400, orange pi 2. after some debugging i found the command what locks

file: Sender.py
command:  913:    self._sumcline = sum(cline)

			print "tosend='%s'"%(repr(tosend)),"stack=",sline,"sum=",sum(cline),"wait=",wait,"pause=",self._pause
			if tosend is not None and sum(cline) < RX_BUFFER_SIZE:
						self._sumcline = sum(cline)  # << this one
#				if isinstance(tosend, list):
#					self.serial.write(str(tosend.pop(0)))
#					if not tosend: tosend = None

				print ">S>",repr(tosend),"stack=",sline,"sum=",sum(cline)
				if self.controller==Utils.SMOOTHIE: tosend = tosend.upper()
				self.serial.write(bytes(tosend))
#				self.serial.write(tosend.encode("utf8"))
#				self.serial.flush()

				tosend = None
				if not self.running and t-tg > G_POLL:
					tosend = b"$G\n"
					sline.append(tosend)
					cline.append(len(tosend))
					tg = t

it produce error in the X server, lock or reboot

after commenting it bCNC began to work without locking/rebooting

bhgv, 2016-05-23:
..continue of debugging

after uncommenting 
```
sender.py:913 -> self._sumcline = sum(cline)
```

next path of debugging:
```
sender.py:913 -> self._sumcline = sum(cline)
--//-- :641 -> def getBufferFill(self): return self._sumcline * 100. / RX_BUFFER_SIZE

bCNC.py:2209  -> self.bufferGauge.setFill(Sender.getBufferFill(self))

lib/tkExtra.py:4475 -> def setFill(self, fill=0.):
--//-- :4480 -> def draw(self, event=None):
--//-- :4485 -> self.itemconfig(self.gaugeArc, extent = -self.fill * 3.6)
```
it looks as the line "self.itemconfig(self.gaugeArc, extent = -self.fill * 3.6)" breaks X after first or one of firsts call. it can be commented freely as it draws only a small round gauge in the status bar. without this command bCNC works well

bhgv, 2016-05-23:
..continue

after some experiments on 
```python
self.itemconfig(self.gaugeArc, extent = -self.fill * 3.6)
```
the result: it breaks X and lock/reboot if extent > abs(285.0). 
i tryed to write 
```python
self.itemconfig(self.gaugeArc, extent = -self.fill * 2.8)
```
and it works for me except a little wrong value showen by this small round gauge.  anyway i think the best is to comment it or replace with textual percent widget

Effer, 2016-05-23:
Hi @bhgv, thanks for your debugging!
I'm embarrassed because I'm the one that added that small round gauge as feedback of the buffer controller. Never tested under linux.
Anyway it can be easily changed in something else, I'll try with colored bar graph later.

Effer, 2016-05-23:
Here another Gauge that doesn't use the tkinter arc.
https://github.com/Effer/bCNC/tree/master-new-gauge

[image]

Now is a linear vertical bar that doesn't change much if the buffer is well filled. If someone could test and see if the problem go away..

bhgv, 2016-05-23:
hi @Effer, thanks to your fast responce.

colored bar from last picture looks very good and more understandable than round gauge (sorry, it's only my opinion). if you make a tooltip on it "buffer filling" or "%% of buffer" it would be excellent.

hmm, 
https://github.com/Effer/bCNC/tree/master-new-gauge 
- this is a link for testing?

bhgv, 2016-05-23:
new gauge works well for me (lubuntu armhf mali400 etc). but maybe you'll do it 1.5-2 times wider? it looks too small and has place only for 2 digits. 
(don't know how to attach here a screenshot or video)

spanner888, 2016-05-23:
New gauge also works well for me. Only had time for one test this morning, but job ran fully with no issues in new code and also confirmed old code still has that repeatable solid logout/crash issue.

A really big thanks!  This has stopped me abandoning bCNC as all my old PCs had this issue and I could not afford any "newer" PCs.

bhgv, 2016-05-24:
@spanner888 this is not a problem of old pc. this is a problem of tkinter lib. this is a internal python lib. you'll receive this problem on any computer with this bugged lib.

vlachoudis, 2016-05-24:
Thanks for figuring out the problem. I don't know if the problem is due to the python lib, however Xorg should not crash from a user program.

bhgv, 2016-05-24:
@vlachoudis yes, you are right, X should nether be crashed by user prog. this bsod happened at first as for tk, as for 2d, as for 3d. and maybe, after more deep debugging it will be possible to tell what it was. anyway it's not bcnc.

but now thanks for your help, super program and super excellent support!

Effer, 2016-05-24:
Here a buffer bar that use a robust widget.
https://github.com/Effer/bCNC/tree/master-Buffer-Bar
Tester are wellcome

bhgv, 2016-05-24:
@Effer 
https://github.com/Effer/bCNC/tree/master-Buffer-Bar
it works fine. without any problems and looks good. tested on pair of gcode scripts that gave bsod. (lubuntu, arm, etc)

PR #332 (2016-05-25): Buffer Bar
To fix #315 buffer fill now use a well tested component.
closed 2016-05-25 by vlachoudis