vlachoudis/bCNC #316: Autolevel

Guenni75, 2016-05-09:
Is there a way to aboard the autolevel Scan? 
I can not find a button for this.

atrueresistance, 2016-05-09:
Abort? You can click the stop button

closed 2016-05-12 by vlachoudis

Guenni75, 2016-05-15:
There is no Stop Button in the Probe Menu.
No Chance to Press Stop at an emergency.

Effer, 2016-05-15:
Hi @Guenni75 . That's true, the default configuration doesn't give you the "Run" ribbon in the probe tab. But you can modify some settings to customize  the tab and give you a configuration like this:

[image]

To do this you have to manually change the configuration file. Unfortunately this have to be done with an external editor. 

_The file is created in you user directory. **NOT** the "bCNC.ini" file in the bCNC folder. The exact location of this file depends on you operating system here for windows:You can reach it easily using the Environment variable %UserProfile% . Hit [Windows key + R] paste %UserProfile% in the Run prompt, then click OK. The user folder will open and there you'll find the .bCNC file. Right click and choose "open with" then select notepad._

Now you have to add this line in [bCNC] section: **probe.ribbon = Connection Probe Run**

The **Run** option will add the buttons you need. Maybe in the future it will be better integrated.

Another customization I like in my setup is to bind some shortcut to the run-stop buttons. In this way I have some physical button to use. Go to Tools-->Shortcut and customize some buttons:

[image]

Response to stop command is fast, but if you need a more reliable and secure way to stop your machine consider to add some real physical buttons, GRBL is ready to accept them.