vlachoudis/bCNC #318: Autolevel Issue Perhaps Z related?

atrueresistance, 2016-05-12:
I think I might have found a new bug in the autolevel. I forgot to save the probe file. Made my first engrave into the aluminum bed :-(

gcode [1-.25adaptive.zip](https://github.com/vlachoudis/bCNC/files/261844/1-.25adaptive.zip)

The board I was cutting was fairly cupped which is why I ran an autolevel on it. In this video you can see where the Z value goes down further then it should resulting in those divots.  The lower side was on the left where the right side was higher.

Video: https://goo.gl/photos/hkT69VXGGbWqaqbi7
[image]

It should look like this. 
[image]

idk perhaps user error, I'm kicking myself for not saving the probe data.

Effer, 2016-05-12:
Maybe G fast Z not interpolated?

vlachoudis, 2016-05-12:
@atrueresistance I think @Effer is right. The G0 is not interpolated. I saw in the code that it uses a G0 to plunge ~1mm above the surface and then uses G1 Z... to start the cut.
I should add the interpolation also on the G0.

atrueresistance, 2016-05-12:
Is that something I need to change in my Fusion360 post?

vlachoudis, 2016-05-13:
It seems that Fusion360 generates a very efficient code, with moves I didn't forecast when I made the autolevel. It would be a pity to change it from Fusion, I've tried to correct it in bCNC. I have not tried it, since I don't have access to my machine right now, so please try to see if it works.

vlachoudis, 2016-05-16:
Did you try the new master?

atrueresistance, 2016-05-16:
@vlachoudis Not yet, tore my machine apart to change out 6mm belting for 9mm belting. Glued together some stock yesterday.

atrueresistance, 2016-05-22:
@vlachoudis  After tearing me machine apart i'm within .009-.005" square in the z. Next time I work on a not even piece and the auto level doesn't work out I'll open an issue.

atrueresistance, 2016-05-22:
it did seems a bit better if that helps prior to upgrade

closed 2016-05-22 by atrueresistance