vlachoudis/bCNC #320: Copy gcode in editor, then paste into external app produces strange results.

spanner888, 2016-05-13:
So if you copy gcode in editor (via menu or Ctl C), then paste into external app produces strange results. A lot of extra text that looks a bit like markers for processing in bCNC code is visible.

Have seen issue in Lubuntu and Windows 7.

Seems to be a low priority issue, as work around is to save gcode via bCNC, then open saved file in external app. Maybe just add a note in FAQ or somewhere in doco?

Zipped example of gcode produced by Bowl tool is attached.
[default_bowl_copy_paste.gcode.zip](https://github.com/vlachoudis/bCNC/files/262411/default_bowl_copy_paste.gcode.zip)

vlachoudis, 2016-05-13:
I am afraid it is normal. It is the internal structure I keep to remember all the features of the gcode. I could add an option to copy the code as text maybe?

Effer, 2016-05-13:
In this way gcode will be exported without all the add stuff? It could be useful.

spanner888, 2016-05-13:
Yes that would be a good option, but def not a high priority as the file
save, already allows gives a way to get the code from saved file.

Effer, 2016-09-16:
@spanner888 you can now save as **G-Code Clean ** from the save dialog.
Only enabled gcode, no bcnc metadata and uppercase.