vlachoudis/bCNC #321: How add submenu Cut Direction (Climb and Conventional) to menu Pocket and Cut?

marketa888, 2016-05-14:


vlachoudis, 2016-05-14:
As it is now, the pocket/profile/cut direction remains as it is defined in the path. However you can change it from the "Editor -> Route" . And you can visualize it with the "Editor -> Info -> Info" or the shortcut "n" in the canvas.

marketa888, 2016-05-14:
I see now , it's easy , thank you.

marketa888, 2016-05-15:
Conventional/Climb/Rewerse in Route-menu not work properly.
It always gives  Climb metod.
I do not know where I make an error?

please see pocket circle 10mm
-----
(Block-name: Header)
(Block-expand: 0)
(Block-enable: 1)
M3 S12000
G4 P3
(Block-name: 0)
(Block-expand: 1)
(Block-enable: 0)
g0 x10 y0
g1 z0 f10
g3 x10 y0 i-10 j0 f127
g0 z1
(Block-name: 0 [pocket,conventional])
(Block-expand: 1)
(Block-enable: 1)
g0 x0.7925 y0
g1 z0 f10
g3 x0.7925 y0 i-0.7925 j0 f127
g1 x2.0625 y0
g3 x2.0625 y0 i-2.0625 j0
g1 x3.3325 y0
g3 x3.3325 y0 i-3.3325 j0
g1 x4.6025 y0
g3 x4.6025 y0 i-4.6025 j0
g1 x5.8725 y0
g3 x5.8725 y0 i-5.8725 j0
g1 x7.1425 y0
g3 x7.1425 y0 i-7.1425 j0
g1 x8.4125 y0
g3 x8.4125 y0 i-8.4125 j0
g0 z1
------
From inside to outside CCW is Climb not Conventional !

and after convert Climb :
-------
(Block-name: Header)
(Block-expand: 0)
(Block-enable: 1)
M3 S12000
G4 P3
(Block-name: 0)
(Block-expand: 1)
(Block-enable: 0)
g0 x10 y0
g1 z0 f10
g3 x10 y0 i-10 j0 f127
g0 z1
(Block-name: 0 [pocket,climb])
(Block-expand: 1)
(Block-enable: 1)
g0 x8.4125 y0
g1 z0 f10
g2 x8.4125 y0 i-8.4125 j0 f127
g1 x7.1425 y0
g2 x7.1425 y0 i-7.1425 j0
g1 x5.8725 y0
g2 x5.8725 y0 i-5.8725 j0
g1 x4.6025 y0
g2 x4.6025 y0 i-4.6025 j0
g1 x3.3325 y0
g2 x3.3325 y0 i-3.3325 j0
g1 x2.0625 y0
g2 x2.0625 y0 i-2.0625 j0
g1 x0.7925 y0
g2 x0.7925 y0 i-0.7925 j0
g0 z1
------
from outside to inside CW it is Climb !
The change applies to the entire block and not just selected gcode block.

vlachoudis, 2016-05-15:
@marketa888 it is true, I forgot to consider the direction in the pocket. I will correct it :)

vlachoudis, 2016-05-15:
@marketa888 actually I was wrong :) the behaviour if you have a g2 and you do a pocket it will create g2, if you have g3 it will create g3, so it keeps the initial orientation

marketa888, 2016-05-15:
OK. And  radius "r"  changes from low to high and vice versa
needlessly. IMHO :(  
I get always climb milling

vlachoudis, 2016-05-16:
Pocket will start always from inside to outside (smaller r to bigger r). This is done on purpose to reduce the amount of cuts that the tool is cutting full diameter.

marketa888, 2016-05-16:
I so want . from small to bigger. If I click reverse selected block I have a radius from large to small :(

marketa888, 2016-05-16:
only change g2g3 not a radius in selected block to change. please

vlachoudis, 2016-05-17:
Toi have to click reverse before the pocket operation

marketa888, 2016-05-18:
So working properly , a little harder to fix later . I miss the icons " find and replace" in the editor.

vlachoudis, 2016-05-19:
Find and replace could be an addition to the editor. Please add it to the "feature request" issue

closed 2016-05-19 by vlachoudis