vlachoudis/bCNC #324: Weird behaviour in Terminal and connect button

ralfsteck, 2016-05-19:
Hi,

I'm unsure where to post my issue because I can't find a forum or another way to ask questions.

I run bCNC on a Protoneer Raspberry Pi CNC board, with their Raspi image. bCNC ran fine in the beginning until I (obviously) hiit a wrong key or button. Now the terminal is flooded with status lines, I can't reaaly work with settings becausxe they dissapear faster than I can read them. This wasn't like that at the start.

Plus the large status button doesn't work anymore. Until the wrong thing I did, the button showed "Connected", "Idle", "Hold", etc. Now there only is a message "not connected" but when I clock on it, it says "bCNC is connected" and yes, I can jog my mill around.

Please, what did I do wrong? Reflashing the Raspi image didn't help.

vlachoudis, 2016-05-19:
Can you send a screenshot of the terminal

ralfsteck, 2016-05-19:
Yes sure :-)
[image]

vlachoudis, 2016-05-19:
Did you recompile grbl yourself and removed the reporting of WPos?
bCNC is requiring the standard message with both WPos and MPos to be reported.

ralfsteck, 2016-05-19:
As I said, this is the standard Raspi image from protoneer (http://wiki.protoneer.co.nz/Raspberry_Pi_CNC_User_Interface_SD_Card_Image[here](url)).

ralfsteck, 2016-05-19:
And I rewrote the SD card (which means all bCNC is as it was when I started) but didn't flash the Arduino. So the problem might more be related to GBRL, but I still hope you can help me.

ralfsteck, 2016-05-19:
I just see that $10 is the status mask which influences exactly that, in the settings of my shapeoko dealer, S10 is arc min.segments. Seems I fiddled there... I'll power up and let you know.

vlachoudis, 2016-05-20:
Ok I will make more flexible the parsing of the status messages to avoid such behaviour

anaronix, 2016-09-30:
I would like to say thank you very much! I was having the exact same issue after installing bCNC under Ubuntu 14.04.2. I am a complete linux newbie and didn't even think this could be a GRBL issue, since I have had bCNC running fine under Windows7. 
I issued the $10=3 statement and immediately bCNC showed 'Connected' and the terminal window stopped being flooded by the <Idle,MPos:0.0000,0.0394,0.0000,Buf:0> message
I did not have that issue when running bCNC under Windows7.
However, Both times I have tried running bCNC under different linux installs I encountered the same issue.
As a side note: I also stumbled with an error message of /dev/tty/USB0 not accessible until I added my user to the dialout group (on both linux install). This may be additional valuable info that should be added to the install wiki?
Thank you for all your excellent work on this project.

matneu27, 2017-02-10:
Hello,

i stumbled over this entry a bit late in this thread, but i use this fine interface just since a couple of days and have the same issue as described but a bit different behaviour. 
I am working on a linux mint box connected to a arduino uno.
In my case it appears every time i connect. I can connect but it doesn’t  switch into idle state. It stays "not connected" and i can't drive any stepper. Also changing the $10 parameters doesn’t do the trick. The only workaround i figured out, was connecting first at 56000 baud, disconnect and then connect at 115200. Now it connects and switches to idle. Then everything works fine.
As a side note, the connection to UGS works without issues. So i assume that the USB connection to the arduino is still fine. 
Let me know if you need additional info. 

I also want to thank you too for this well done and pretty interface

vlachoudis, 2017-02-11:
@matneu27 when grbl 0.9 is used bCNC expects both the MPos and WPos, since it needs that to find the offset to display the gantry position on canvas. All these problems are solved with the grbl v1.0+

closed 2017-02-11 by vlachoudis