vlachoudis/bCNC #325: Spindle RPM translation

ralfsteck, 2016-05-19:
Hi,

I run a Shapeoko-like mill with a china PWM spindle. The Protoneer wiki says the PWM values are between 0 and 1000 which means 0% to 100% spindle speed. I'm aware that I can change spindle max to 1000, which means that I can use the whole way of the slider now (initially it reached to 12000, but full spindle speed was at 1000).

The spindle can run at 3000-12000rpm and I would prefer to have those values in the UI. So I seek something like a translation factor setting so I could set 200 (the minimum my spindle needs to run) = 3000rpm, 1000=12000rpm.

I'm aware that this has nothing to do with real rpm, but it would be an approximation (at least better than the 1000 I have now).

vlachoudis, 2016-05-19:
In Tools->Machine you can set the min/max RPM value of the spindle. But these values are used for the display of the slider bar and are send literally to grbl, which also should be configured accordingly.

ralfsteck, 2016-05-19:
I know that, I wish there would be a translation so that the UI sais 12000 and bCNC reports 1000 to GBRL.

BlueOrangeLive, 2016-05-19:
@ralfsteck open "Terminal" in bCNC;
$$
$ 30 = 24000 (rpm max)
$ 31 = 0 (rpm min)
Enter:
$30=12000
$31=3000
$$
Or in grbl customize the file default.h
Greeting Jürgen
- - - - -
@ralfsteck in "bCNC" Terminal öffnen ; 
$$  
$30=24000 (rpm max)
$31=0    (rpm min)
Eingeben:
$30=12000 
$31=3000
$$
Oder in grbl die Datei default.h anpassen
Gruß Jürgen

ralfsteck, 2016-05-19:
Seems my GBRL doesn't know about G30/G31 :-(

BlueOrangeLive, 2016-05-19:
This is not a G-Code.
Connect with bCNC grbl and then up right in bCNC Click on Terminal and enter the bottom left command:  $$.
:-)

BlueOrangeLive, 2016-05-19:
[image]
[image]

ralfsteck, 2016-05-20:
Yes i know. That was just a typo, sorry $30 and $31 don't exist here.
[image]

vlachoudis, 2016-05-20:
It seems you have an old version of grbl, you have to update it

BlueOrangeLive, 2016-05-20:
@vlachoudis I think the same.
@ralfsteck 
OK. Ich verwende einen Aktuellen Download von bCNC, Oben rechts in Tools System Deutsch auswählen und neustarten. Ist seit kurzem dabei.
Zu grbl benutzt du denke ich die  VERSION "0.9j" aus dem Master da ist das auch noch nicht so. Dort kann man in der Datei config.h in Zeile 
254:  #define SPINDLE_MAX_RPM 1000.0
255: #define SPINDLE_MIN_RPM 0.0
die eigenen Werte anpassen und dann neu Kompelieren bzw in der Arduino IDE aufspielen.
Ich nutze das edge repositie da ist das schon drin (VERSION "1.0c).

ralfsteck, 2016-05-20:
Kann ich denn den Arduino Nano aus dem Raspberry PiCNC ausbauen und dann am PC an USB anschließen und flashen? Ich habe das bisher nur bei den großen Arduinos gemacht bei meinem 3D-Drucker.

ralfsteck, 2016-05-20:
Other question: I love you for helping me but this here is not the place for such things. Would't it be better to have a forum or a Google group for this?

BlueOrangeLive, 2016-05-20:
Ja kannst du, den Nano auswählen. dann das grbl aus edge auspacken. Unter /examples/grblUpload die Datei grblUpload.ino nutzen. Aber zuerst must du grbl als library in Arduino einfügen.

ralfsteck, 2016-05-20:
Jürgen, that worked. I changed config.h and uploaded it to the Nano. Now the rpms work right.

closed 2016-05-27 by vlachoudis

Harvie, 2020-11-23:
Now there seems to be quite elegant solution integrating Huanyang RS485 communication directly to GRBL, i would be happy if you can help with testing:
https://github.com/terjeio/grblHAL/issues/68