vlachoudis/bCNC #326: Wirering raspbery pi 3 (bcnc) to arduino (grbl)

bjao1, 2016-05-20:
I loaded bcnc to raspberry pi3 & grbl to arduino uno.
connected them using usb cable
Problem is:
I can not get them to communicate.
The serial port is not available in bcnc.
Do I connect the tx-rx pins of both boards?
Please help
Thanks

lalo-uy, 2016-05-20:
NO, since you conected them with USB cable, in bCNC look for USB serial
ports like /dev/ttyUSB0
may be a raspberry user have no access to them, try "sudo bCNC"

ralfsteck, 2016-05-20:
On my Pis, those ports hve Names like /dev/ACM0 or /dev/AMA0. Look for those.

bjao1, 2016-05-20:
I  installed the bCNC image from  http://wiki.protoneer.co.nz/Raspberry_Pi_CNC_User_Interface_SD_Card_Image
Using file manager, I see the dev/AMAO file, but in bCNC the only option under connection icon is Hard Reset. I run the RPi3 Serial config according to the instructions. Still no serial port.
I am new to Raspberry pi and Linux, so please bear with me.
Thanks for your help

Protoneer, 2016-05-20:
The serial config in the instructions are related to the hardware serial
and should not effect USB devices.

Might be worth trying to connect to your USB device via a terminal app like
Minicom.

bjao1, 2016-05-20:
Yes, I used minicom and it still could not fine a serial port

bjao1, 2016-05-20:
Can I connect the tx-rx pins of both boards to make it work?I just want them to communicate so I can wire the arduino uno grbl to my stepper driver
Thanks for your helpRobert

lalo-uy, 2016-05-20:
Raspberry Pi is a 3V device and Arduino UNO is 5V device, you may damage
the Raspberry

bjao1, 2016-05-20:
if this is the way to go, I can make the txd & rxd pass through a voltage translator. 
how is your setup with bCNC RPi3 and grbl arduino uno? I mean how are they connected?
Thanks

nfz, 2016-05-21:
I use a RPi2 and RPi3 with USB to grbl with no problems.  You may have to setup your user as a member of the dialout group.

Assuming you are using the default login then for user pi:

`sudo usermod -a -G dialout pi`

bjao1, 2016-05-22:
Hi nfz
I tried your suggestion, still not change (can  not communicate with arduino grbl
Are you using bCNC image from protoneer wiki site? If not can you share what bCNC you are using?
Thanks

nfz, 2016-05-22:
I should have mentioned you would probably have to reboot your pi after issuing the usermod command as it would not take effect until the next time you logged in.

I use bCNC from this github, usually do a daily git pull to stay up to date with the latest changes.

Also make sure pyserial is installed but if you are using the protoneer's image then that should already be setup.

To make sure that the arduino board is actually powering up when you plug in the USB cable, do
`lsusb`
before plugging the board into the RPi3 and then after.  List the results here.  lsusb will list all the active USB devices plugged into your RPi3.

Then do:
`dmesg`
after you have plugged the arduino into the RPi3.
We are only interested in the last 5 lines and list those here. dmesg will list all device messages in the kernal ring buffer and we are interested in the device messages for the USB driver that gets setup when the arduino gets plugged into the RPi3.

Here are the last few lines of dmesg output on my RPi3 after I plug in the arduino board running grbl:
[image]

lalo-uy, 2016-05-22:
can you post the result of diss comand in the PI:

ls -l /dev

bjao1, 2016-05-22:
lsusb before and after

pi@raspberrypi:~ $ lsusb
Bus 001 Device 004: ID 0101:0007  
Bus 001 Device 006: ID 1a2c:0e24 China Resource Semico Co., Ltd 
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@raspberrypi:~ $ lsusb
Bus 001 Device 007: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
Bus 001 Device 004: ID 0101:0007  
Bus 001 Device 006: ID 1a2c:0e24 China Resource Semico Co., Ltd 
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@raspberrypi:~ $

bjao1, 2016-05-22:
pi@raspberrypi:~ $ ls -l /dev
total 0
crw------- 1 root root     10, 235 May 22 14:17 autofs
drwxr-xr-x 2 root root         580 May 22 14:17 block
crw------- 1 root root     10, 234 May 22 14:17 btrfs-control
drwxr-xr-x 3 root root          60 Jan  1  1970 bus
crw------- 1 root root     10,  63 May 22 14:17 cachefiles
drwxr-xr-x 2 root root        2640 May 22 15:00 char
crw------- 1 root root      5,   1 May 22 14:17 console
crw------- 1 root root     10,  62 May 22 14:17 cpu_dma_latency
crw------- 1 root root     10, 203 May 22 14:17 cuse
drwxr-xr-x 6 root root         120 May 22 14:17 disk
crw-rw---- 1 root video    29,   0 May 22 14:17 fb0
lrwxrwxrwx 1 root root          13 Jan  1  1970 fd -> /proc/self/fd
crw-rw-rw- 1 root root      1,   7 May 22 14:17 full
crw-rw-rw- 1 root root     10, 229 May 22 14:17 fuse
crw-rw---- 1 root gpio    244,   0 May 22 14:17 gpiomem
crw------- 1 root root    247,   0 May 22 14:17 hidraw0
crw------- 1 root root    247,   1 May 22 14:17 hidraw1
crw------- 1 root root    247,   2 May 22 14:17 hidraw2
crw------- 1 root root     10, 183 May 22 14:17 hwrng
lrwxrwxrwx 1 root root          25 Jan  1  1970 initctl -> /run/systemd/initctl/fifo
drwxr-xr-x 4 root root         180 May 22 14:17 input
crw-r--r-- 1 root root      1,  11 May 22 14:17 kmsg
lrwxrwxrwx 1 root root          28 Jan  1  1970 log -> /run/systemd/journal/dev-log
brw-rw---- 1 root disk      7,   0 May 22 14:17 loop0
brw-rw---- 1 root disk      7,   1 May 22 14:17 loop1
brw-rw---- 1 root disk      7,   2 May 22 14:17 loop2
brw-rw---- 1 root disk      7,   3 May 22 14:17 loop3
brw-rw---- 1 root disk      7,   4 May 22 14:17 loop4
brw-rw---- 1 root disk      7,   5 May 22 14:17 loop5
brw-rw---- 1 root disk      7,   6 May 22 14:17 loop6
brw-rw---- 1 root disk      7,   7 May 22 14:17 loop7
crw-rw---- 1 root disk     10, 237 May 22 14:17 loop-control
drwxr-xr-x 2 root root          60 May 22 14:17 mapper
crw-r----- 1 root kmem      1,   1 May 22 14:17 mem
crw------- 1 root root     10,  59 May 22 14:17 memory_bandwidth
brw-rw---- 1 root disk    179,   0 May 22 14:17 mmcblk0
brw-rw---- 1 root disk    179,   1 May 22 14:17 mmcblk0p1
brw-rw---- 1 root disk    179,   2 May 22 14:17 mmcblk0p2
drwxrwxrwt 2 root root          40 Jan  1  1970 mqueue
drwxr-xr-x 2 root root          60 May 22 14:17 net
crw------- 1 root root     10,  61 May 22 14:17 network_latency
crw------- 1 root root     10,  60 May 22 14:17 network_throughput
crw-rw-rw- 1 root root      1,   3 May 22 14:17 null
crw------- 1 root root    108,   0 May 22 14:17 ppp
crw-rw-rw- 1 root tty       5,   2 May 22 15:04 ptmx
drwxr-xr-x 2 root root           0 Jan  1  1970 pts
brw-rw---- 1 root disk      1,   0 May 22 14:17 ram0
brw-rw---- 1 root disk      1,   1 May 22 14:17 ram1
brw-rw---- 1 root disk      1,  10 May 22 14:17 ram10
brw-rw---- 1 root disk      1,  11 May 22 14:17 ram11
brw-rw---- 1 root disk      1,  12 May 22 14:17 ram12
brw-rw---- 1 root disk      1,  13 May 22 14:17 ram13
brw-rw---- 1 root disk      1,  14 May 22 14:17 ram14
brw-rw---- 1 root disk      1,  15 May 22 14:17 ram15
brw-rw---- 1 root disk      1,   2 May 22 14:17 ram2
brw-rw---- 1 root disk      1,   3 May 22 14:17 ram3
brw-rw---- 1 root disk      1,   4 May 22 14:17 ram4
brw-rw---- 1 root disk      1,   5 May 22 14:17 ram5
brw-rw---- 1 root disk      1,   6 May 22 14:17 ram6
brw-rw---- 1 root disk      1,   7 May 22 14:17 ram7
brw-rw---- 1 root disk      1,   8 May 22 14:17 ram8
brw-rw---- 1 root disk      1,   9 May 22 14:17 ram9
crw-rw-rw- 1 root root      1,   8 May 22 14:17 random
drwxr-xr-x 2 root root          60 Jan  1  1970 raw
crw-rw-r-- 1 root root     10,  58 May 22 14:17 rfkill
drwxr-xr-x 4 root root          80 May 22 15:00 serial
lrwxrwxrwx 1 root root           7 May 22 14:17 serial0 -> ttyAMA0
lrwxrwxrwx 1 root root           5 May 22 14:17 serial1 -> ttyS0
[... 77 lines cut ...]
crw------- 1 root root    251,   0 May 22 14:17 vc-cma
crw-rw---- 1 root video   248,   0 May 22 14:17 vchiq
crw-rw---- 1 root video   249,   0 May 22 14:17 vcio
crw------- 1 root root    250,   0 May 22 14:17 vc-mem
crw-rw---- 1 root tty       7,   0 May 22 14:17 vcs
crw-rw---- 1 root tty       7,   1 May 22 14:17 vcs1
crw-rw---- 1 root tty       7,   2 May 22 14:17 vcs2
crw-rw---- 1 root tty       7,   3 May 22 14:17 vcs3
crw-rw---- 1 root tty       7,   4 May 22 14:17 vcs4
crw-rw---- 1 root tty       7,   5 May 22 14:17 vcs5
crw-rw---- 1 root tty       7,   6 May 22 14:17 vcs6
crw-rw---- 1 root tty       7,   7 May 22 14:17 vcs7
crw-rw---- 1 root tty       7, 128 May 22 14:17 vcsa
crw-rw---- 1 root tty       7, 129 May 22 14:17 vcsa1
crw-rw---- 1 root tty       7, 130 May 22 14:17 vcsa2
crw-rw---- 1 root tty       7, 131 May 22 14:17 vcsa3
crw-rw---- 1 root tty       7, 132 May 22 14:17 vcsa4
crw-rw---- 1 root tty       7, 133 May 22 14:17 vcsa5
crw-rw---- 1 root tty       7, 134 May 22 14:17 vcsa6
crw-rw---- 1 root tty       7, 135 May 22 14:17 vcsa7
crw-rw---- 1 root video   246,   0 May 22 14:17 vcsm
crw------- 1 root root     10, 137 May 22 14:17 vhci
prw-r----- 1 root adm            0 May 22 15:03 xconsole
crw-rw-rw- 1 root root      1,   5 May 22 14:17 zero
pi@raspberrypi:~ $

nfz, 2016-05-22:
Looks like your arduino board is using a HL-340 USB serial chip and its connected to /dev/ttyUSB0 and belongs to the dialout group.  You should be able to connect.  You may have to type in `/dev/ttyUSB0` in the Port setting although the drop down list should show it.
[image]

After changing the port you will have to left click on the open icon to the right.

bjao1, 2016-05-22:
HI NFZ
A million thanks for your help!
I is ALIVE!!!!!

nfz, 2016-05-22:
Now the real fun begins.

closed 2016-05-23 by vlachoudis