vlachoudis/bCNC #335: OpenCV

Guenni75, 2016-05-25:
Hi.
I have a problem with the Camera.
I can't get it to work.
I downloaded a lot of stuff but nothing seems to work.
You wrote in the wiki:
"note: Be sure you have also installed PIL, OpenCv, ImageTk for your Python."
I installed PIL, ImageTK and downloaded OpenCV (just an archiv, where must the files be moved?)...nothing works. Still a grey Button!

Any Ideas?

vlachoudis, 2016-05-25:
Can you see if any message is written on the terminal?
Otherwise you can run python your self from a terminal and type the 2 import commands. To see which import is failing
```
$ python
>>> import cv2
>>> import PIL.ImageTk
```

Effer, 2016-05-25:
Hi @Guenni75 , 
if you are under windows: 
- install Numpy (prerequisite) from https://sourceforge.net/projects/numpy/files/NumPy/1.7.1/numpy-1.7.1-win32-superpack-python2.7.exe/download  
- Extract OpenCV to c:/opencv
- Goto c:/opencv/build/python/2.7 folder and copy the file cv2.pyd to c:/Python27/lib/site-packages.
- Then test if it works

vlachoudis, 2016-05-27:
@Guenni75 did you succeed?

Guenni75, 2016-05-29:
I have tested it yet.
Installed Numpy and copied opencv into the folder.

Still grey Camera Button.

[image]

I don't understand the error message, but i think there is something missing!

Guenni75, 2016-05-29:
Sorry

Effer, 2016-05-29:
Hi @Guenni75 , it's seems like neither PIL neither opencv are correctly installed. 

For Pillow (PIL) I use pip that is already installed from python version 2.7.9 (in windows)
From command prompt type **pip install Pillow** .
If windows can't find pip you can navigate to c:\Python27\Scripts from the prompt and then execute the command above.  This should fix the problem with PIL.

Now for opencv:
in the above instruction this is my c:\opencv folder

[image]

then I've copied the **cv2.pyd** as the instruction above and python get the opecv support. Maybe a reboot is needed but nothing more. Which version do you download? I'm using 2.4.11 FYI

Guenni75, 2016-05-30:
Hi.
"pip install Pillow" has an SyntaxError: Invalid Syntax

How can i change the Dir? 
I don't know the commands of Python. Everything i found in the Net doesn't work.

Guenni75, 2016-05-30:
Ok, i made it.
Nobody said that i have to jump between Python Terminal and WIndows Terminal.
I changed the Directorys in WIndows and could start the installs in Python.
Then i had error messages i could work with.
I updated Numpy and the installation of cv2 works.

closed 2016-05-30 by Guenni75