vlachoudis/bCNC #336: Camera for monitoring

ralfsteck, 2016-05-25:
Hi,

I would love to se an (optional?) Tab where I can see a camera picture. So I could install a USB camera and watch if everything is going right without being at the mill.

Might not be difficult with mjpeg-streamer and a window where the stream is displayed (that's how the people at Octoprint do it für 3D printers), but my programming skills are Zero.

vlachoudis, 2016-05-25:
If you install OpenCV for python, you can access already any webcam that recognized by opencv in two ways:
- Canvas: Search for the "camera-icon" inside the Canvas to activate it, and in the Tab "Probe->Camera" you can adjust the position of the camera.
- Web pendant: if you activate the web pendant from "File-> Pendant" you can access the page
  localhost/camera.html and you will get live updates every 1s on your phone/tablet etc.

The cameras can be configured for the moment partially in the "Tools->Config->Camera" but you can modify everything from the ~/.bCNC configuration file

Effer, 2016-05-25:
If I'm not wrong the default address for the web pendant is on port 8080.
So in my case the camera page is at [http://localhost:8080/camera.html](http://localhost:8080/camera.html)

closed 2016-05-27 by vlachoudis