vlachoudis/bCNC #344: Probe Camera Alignment

Guenni75, 2016-06-01:
Hi.
I tested the Probe Camera Alignment.
I made the setup with the wiki, but when i press the "Switch to" Button the Spindel moves in the opposite direction. Only Z is ok.
When i make the negativ X and Y positiv, it is ok.

vlachoudis, 2016-06-03:
The idea with the "Switch to" button is to switch between the camera and spindle. Once you calibrate the camera, and you click the button it will move the camera to the spindle location the first time and the inverse the second.

vlachoudis, 2016-06-07:
@Guenni75 also please note that the switch between camera and spindle is performed by setting the relative offset coordinates G90

Guenni75, 2016-06-08:
The Camera is nearly at the same Position as in the wiki (left side of the spindle).
After Drilling the Hole, i move X and Y positiv (just like the wiki), but after pressing "2.Camera" the Offset is negativ. The Values in the wiki are positiv.

If I press "Switch to", it moves the Spindel not down/left, it moves up/right!
If I press "Switch to" again it moves the camera back to the hole.
If I make the negativ Offset positiv, it works.

Could it be, that the "Switch to" Button was "pressed" befor the calibration?
Then i could understand the negativ values.

vlachoudis, 2016-06-08:
That could be a possible explanation or if the G92 was not zero before the calibration

closed 2016-06-19 by vlachoudis