vlachoudis/bCNC #345: DRO decimal place stting

tklus, 2016-06-02:
Is there a way to set the decimal place to 0.000 or 0.0000 all the time. for example, if the read out is displaying .5, I would like it to read 0.500 or 0.5000, if it is displaying .25 to read .250 or .2500. if it is reading 0 it would display 0.000

the DRO currently drops the trailing zeros.

It's not a huge deal but was curious if there was a setting for this.

Thanks,
Tim

Effer, 2016-06-02:
Hi @tklus ,
the option is missing right now. It can be easy added, I can give it a try next days. I like the suggestion.

tklus, 2016-06-04:
Nice work guys! The new setting works great! Should it be defaulted as 3 or 4? At the current 0, when you jog the machine it looks like it is not moving on the DRO because it is rounding to 1.

I would think most users would want either .000 or .0000 

Thanks again guys for the awesome support! I foresee this interface being very popular!

Effer, 2016-06-05:
@tklus just PR to fix that. Thanks to point it out

closed 2016-06-05 by vlachoudis