vlachoudis/bCNC #346: lasercutter setting

tklus, 2016-06-03:
Curious what the lasercutter setting in the machine configuration does. I am currently running a version of GRBL that is optimized for laser engravers. The version of GRBL I am running removes the pause for spindle on and speed changes to eliminate delay on the laser engraver on and off commands.

vlachoudis, 2016-06-03:
When the lasercutter is enabled, the generated gcode (importing from dxf or from internal tools), instead of generating motion in the z-plane to enter and retract from the material it issues M3/M5 commands to turn on/off the laser.

tklus, 2016-06-03:
Perfect! This could be useful for programming raw DXF files within bcnc.
I will have to try it out.
Another issue I noticed in the window that draws the toolpath, on my setup, the traced path was not a nice smooth curve, it was severely faceted. the tool follows the curve but the displayed toolpath line was not smooth. is there a setting for that?

closed 2016-06-04 by tklus

vlachoudis, 2016-06-04:
There is a precision on the maximum allowable sagitta when drawing in the preferences. However the default value should be small enough not to see the facets. Could you post your gcode

tklus, 2016-06-04:
It seems to display fine on my main windows PC but on the raspberry pi and my PC at work it displays like the attached image

[image]
%
(T5  D=0.125 CR=0 - ZMIN=-0.628 - flat end mill)
G90 G94
G17
G20
G28 G91 Z0
G90

(2D Contour1 2)
M9
T5 M6
S10000 M3
G54
M8
G0 X0.7612 Y2.1281
Z0.6
Z0.2
G1 Z0.0794 F40
Z-0.0275
G18 G2 X0.7737 Z-0.04 I0.0125
G1 X0.7862
G17 G3 X0.7987 Y2.1406 J0.0125
G1 Y2.3834 F30
G3 X0.7896 Y2.3938 I-0.0105
G1 X0.4296 Y2.4444
G3 X0.418 Y2.4366 I-0.0015 J-0.0104
G2 X0.2744 Y2.34 I-0.1256 J0.0317
G1 X0.1899 Y2.3519
G2 X0.0804 Y2.5028 I0.018 J0.1282
X1.7478 Y4.1701 I2.0277 J-0.3604
X1.8987 Y4.0607 I0.0227 J-0.1275
G1 X1.9099 Y3.9804
G2 X1.8134 Y3.8368 I-0.1282 J-0.018
G3 X1.8055 Y3.8252 I0.0026 J-0.0102
G1 X1.8567 Y3.4609
G3 X1.8671 Y3.4519 I0.0104 J0.0015
G1 X2.3492
G3 X2.3596 Y3.4609 J0.0105
G1 X2.4102 Y3.8209
G3 X2.4024 Y3.8326 I-0.0104 J0.0015
G2 X2.3058 Y3.9761 I0.0317 J0.1256
G1 X2.3177 Y4.0606
G2 X2.4686 Y4.1701 I0.1282 J-0.018
X4.1359 Y2.5028 I-0.3604 J-2.0277
X4.0264 Y2.3519 I-0.1275 J-0.0227
G1 X3.9462 Y2.3406
G2 X3.8026 Y2.4372 I-0.018 J0.1282
G3 X3.7909 Y2.445 I-0.0102 J-0.0026
G1 X3.4267 Y2.3938
G3 X3.4177 Y2.3834 I0.0015 J-0.0104
G1 Y1.9013
G3 X3.4267 Y1.8909 I0.0105
G1 X3.7867 Y1.8403
G3 X3.7983 Y1.8482 I0.0015 J0.0104
G2 X3.9419 Y1.9448 I0.1256 J-0.0317
G1 X4.0264 Y1.9329
G2 X4.1359 Y1.782 I-0.018 J-0.1282
X2.4686 Y0.1147 I-2.0277 J0.3604
X2.3177 Y0.2242 I-0.0227 J0.1275
G1 X2.3058 Y0.3086
G2 X2.4024 Y0.4522 I0.1282 J0.018
G3 X2.4102 Y0.4639 I-0.0026 J0.0102
G1 X2.3596 Y0.8239
G3 X2.3492 Y0.8329 I-0.0104 J-0.0015
G1 X1.8671
G3 X1.8567 Y0.8239 J-0.0105
G1 X1.8055 Y0.4596
[... 1003 lines cut ...]
G3 X1.8567 Y0.8239 J-0.0105
G1 X1.8055 Y0.4596
G3 X1.8133 Y0.448 I0.0104 J-0.0015
G2 X1.9099 Y0.3044 I-0.0317 J-0.1256
G1 X1.8987 Y0.2241
G2 X1.7478 Y0.1147 I-0.1282 J0.018
X0.0804 Y1.782 I0.3604 J2.0277
X0.1899 Y1.9329 I0.1275 J0.0227
G1 X0.2701 Y1.9442
G2 X0.4137 Y1.8476 I0.018 J-0.1282
G3 X0.4254 Y1.8397 I0.0102 J0.0026
G1 X0.7896 Y1.8909
G3 X0.7987 Y1.9013 I-0.0015 J0.0104
G1 Y2.1406
G3 X0.7862 Y2.1531 I-0.0125 F40
G1 X0.7737
G18 G3 X0.7612 Z-0.6155 K0.0125
G0 Z0.6
G17
M9
G28 G91 Z0
G28 X0 Y0
M5
M30
%