vlachoudis/bCNC #347: Raspberry pi DRO text window not resizing

tklus, 2016-06-04:
I am running bcnc on my raspberry pi 2 and when I increase the size of the font for the DRO the window for the text does not increase. the font is basically cut in half and does not display the full size. its like the pane for the control needs to be increased in size vertically but it won't let me grab the edge to resize.

any ideas?

Thanks,
Tim

Effer, 2016-06-04:
Hi @tklus , I've made a test in windows and it resize correctly.
Here at 48pt font
[image]

Which font are you using? At which resolution is set your monitor?

tklus, 2016-06-04:
I will check screen resolution. I am using a digital ttf font i downloaded and installed. I will try a standard font from the Rpi.

tklus, 2016-06-04:
Thanks for the quick reply! I changed the font to a standard rpi font and it resizes just fine. I tried typing the digital font in the typing program on the pi and it gets chopped up in there too. no biggie to run a different font.

Thanks for the help!
Tim

Effer, 2016-06-04:
You are welcome. Feel free to close the issue.

closed 2016-06-04 by tklus