vlachoudis/bCNC #348: [request]

atrueresistance, 2016-06-04:
Is there a way to be able to send a command much like the file footer, but instead a json or webapi hit? I have something like [this](http://www.amazon.com/Securifi-Remotely-Control-Appliances-Interface/dp/B00TC9NC82) (the older version) which connects to my router. This is plugged into the tool port on my vacuum. I can turn the router on my SO3 on/off with my phone or logged into the router. There is an api, and it would be cool to get it working at the end to shut the router down by triggering an api call.

Effer, 2016-06-04:
With the integrated python script parse, you could do quite everything. You have simply to proceed the python code with the **%** symbol.
Here I call a external program (saved in c:\beep.exe) with a user button.

[image]

In tehory the same thing can be done even in the gCode, but there are some drawback.
[image]
First you must proceed the code with the **%wait ** command to synchronize the stream with the controller.
But the big problem is that the code is executed every time you modify the line in the editor or reload the file.
Maybe this behavior should be changed.

vlachoudis, 2016-06-07:
@Effer it is executed on every refresh since it expects to return a dynamic gcode.
Maybe we can have a keyword as a prefix to specify when to execute the code

Effer, 2016-06-07:
@vlachoudis yes, something that could identify what to process only when it is executed in grbl. This to avoid even the %wait command. In this way it could be easy even to use the RPI GPIO.

vlachoudis, 2016-06-09:
@Effer I've added your request in #310

closed 2016-06-09 by vlachoudis