vlachoudis/bCNC #353: ~feature request~ Uploaded Gcode window

tklus, 2016-06-06:
Would it be possible to have a window below the jog buttons that shows the uploaded Gcode. it could probably just be an expansion of the command line. When I upload the gcode I like to scan down the first bit of code to make sure all is well. Or, disable the graphical display of the machine moves and just show the gcode.

I was also having issues manually entering commands to GRBL in the command line. not sure if it's just my setup or what. I am currently running GRBL .9g set up by Jtech for laser etching. I can't wait until "laser mode" is made available as a setting in GRBL.

Thanks for the help!
Tim

vlachoudis, 2016-06-06:
Theoretically you should be able to do already. just copy the line of the
control.page = 
from the bCNC.ini file to your .bCNC  configuration in the [bCNC] section
and append the word "Terminal*"

it should include the terminal there.
However I fear that there will be a clash with the Terminal page, so you better remove the terminal there.
But I never tried it if it works :)

tklus, 2016-06-06:
I guess I need to play with customizing it.

Would it be possible to have a version of bcnc that is stripped down for running on the raspberry pi? without the visualizer and other plug ins. 
 like what they have done with universal gcode sender.

[image]

vlachoudis, 2016-06-07:
The additional plugins don't really matter if they are there or not.
The visualizer you can always disable.
However as it is now you cannot have the Terminal window in the place of the visualizer.

closed 2016-06-09 by vlachoudis