vlachoudis/bCNC #354: drill and etch  not aligned

tommyken, 2016-06-06:
Hello,
Im using cyclone , bcnc (on rpi2) and pcbgcode for eagle.

I created a drill and etch nc files with the ULP program pcbgcode. 
In BCNC, i did firste probe and autolevel (i found it in tutorial). After this step I opened and aligned the drill file to left corner. After i drilled te holes i changend the drill tot de milling bit with probe and autozero. As last step i did  same way tot the etch file as the drill. But de drill holes doesn't match the positions when i mills the etch file. Its not just a mm, but around 5 mm. Anyone experienced this problem or issue ?

vlachoudis, 2016-06-07:
You don't need to re-align the bit on the left-corner in the etching step.

vlachoudis, 2016-06-12:
DId you succeed? I've just made a pcb with pcbgcode and everything was aligned properly.

tommyken, 2016-06-12:
Actually no, but i found the problem. I have to apply a border in EAGLE. But now it mills when in drill file. And also drills in the etch(mill) file. Its too difficult to me. I tried to find the right tutorial, but cant find it and  the standards doesn't work. Without border the don't  'match'. Sorry for it

vlachoudis, 2016-06-12:
Could you post your etch and drill files that have the problem to check.

tommyken, 2016-06-12:
https://www.dropbox.com/s/0gnyernizcbgl9a/ledstrip.bot.drill.tap?dl=0
https://www.dropbox.com/s/27spldrjy7weobk/ledstrip.bot.etch.tap?dl=0

This was my latest attempt. I tried so many that i don't know which are the best settings.

vlachoudis, 2016-06-13:
The drill position and the etch file are perfectly aligned. You can already see it if you open the etc file and then from Editor-> Edit -> Import
import the drill file. The drill locations are perfectly centered.
Somehow you destroy the alignment when you process one file and you switch to the second. Do you have homing switches?

tommyken, 2016-06-13:
strange, when i use BCNC i got this, in step 3 i see there's no space between the border and drillings
[image]
[image]
[image]

Currently i'm using 0.9.1, might tnhis be the problem ?

vlachoudis, 2016-06-13:
The "magenta" rectangle shows the margins of the gcode that is enabled, and not the borders of the pcb. So the etch one has to be bigger than the drill one. You should not align to the bottom-left corner. Actually you **should not** move the gcode at all.
Also for information pcbgcode generates the gcode in the negative space so the top-right corner should be close to 0,0 

- Load the etch file or the drill
- Place the pcb on the worksurface move the gantry to the whatever corner you want to align, e.g. the bottom left corner of the pcb.
- Then click on the "Set WPOS" button and click on the canvas the same location that corresponds to the bottom-left corner
- From this point one you should not change the WPos or move your gcode. 
- Mill the etch file
- then load the drill file (and again you don't move change anymore WPos)
- now the drill should be perfectly aligned with the previous one

Alternatively if you want to move it as you are doing now (**not recommended at all**)
- load the etch file
- move the code with the B-L button
- on the bottom of the screen it will write how much it moved the gcode. Write somewhere this command
- mill your etch
- once you load the drill file. Manually type the "MOVE ###  ### ###" command you wrote to move the drill to the exact same position.

PS. I don't know what is better first the etch and then the drill or vise versa? For sure the cut should be last :)

vlachoudis, 2016-06-19:
@tommyken  did you succeed? I am closing the issue if you don't succeed please reopen again

closed 2016-06-19 by vlachoudis

tommyken, 2016-06-19:
No i cant find the solution, except the not recommended solution (the one with MOVE). WPOS and more i don't understand.I was hoping there was an easy way.
[image]

vlachoudis, 2016-06-23:
@tommyken it could well be that the problem was the same as in #366 metric/imperial conversion. Apparently the drilling generated from pcbgcode doesn't have the G20 in the header. Maybe the fix I've made for #366 works also for you.