vlachoudis/bCNC #355: error: bad number format

UweSeifert, 2016-06-07:
First I load an g-code file. Then I go to the editor tab select all objects and move them to a new location. After the move all positions are updated correctly but also the first G1 line of my gcode file. See the first three lines before the move :   

G21         ; Set units to mm
G90         ; Absolute positioning
G1 Z2 F200  ;Movetoclearancelevel

and after the move. 

G21         ; Set units to mm
G90         ; Absolute positioning
G1 Z2 F0 Movetoclearancelevel

when I correct the last line manualy all is fine.

vlachoudis, 2016-06-07:
Thanks for pointing it out. The ";" comment was missing from the parsing for transformations. Now it is corrected

closed 2016-06-07 by vlachoudis

UweSeifert, 2016-06-07:
not only the ; was missing. Also the feed value was wrong. The original value was F200. After the movement it was F0.

UweSeifert, 2016-06-07:
I checked your change. Now the G21 command is correct but the comment is deleted.. No problem with that. it's working now. Thanks.

vlachoudis, 2016-06-07:
I will try in the future to retain the inline comments :)