vlachoudis/bCNC #356: ring center finder problem

levis0, 2016-06-08:
I try to use ring center finder and after probe reaches left edge i got PROBE: error and bCNC stops working at all. I need to restart it. Z axis zeroing and autolever is working fine. I think here is a problem with used G-code algorythm - then executed first G38 code ir moves probe until it got contact with surface, but after, it leaves in contact and second g-code G38 got error, because probe allready is in contact with ring and cannot be executed. i'm using grbl 0.8i. or i'm doing something wrong ?

levis0, 2016-06-08:
sorry grbl is 0.9i

BlueOrangeLive, 2016-06-11:
Can I confirm here. Same for me.
Grbl vrsion: 1.0d.20160410 on Mega2560 and bCNC of Today.
Even if an input to small diameter, it can not reset the error message. Restarting bCNC is required.

Probe Feed: change is adopted only after a restart.

To my homemade sensor for testing, I have for me in the file: Inserted ProbePage.py two lines, see Picture

lines.append ("g53 g0 x [(prbx + 2)]")
and
lines.append ("g53 g0 y [(prby + 2)]")
[image]

BlueOrangeLive, 2016-06-11:
Here's a little video from the Center test my probe.

https://www.youtube.com/watch?v=cDmEZm-dOj4

vlachoudis, 2016-06-12:
Thanks both. I've correct it by moving by diameter/10. Starngely enough with the previous version of grbl it was working ok.

closed 2016-06-12 by vlachoudis