vlachoudis/bCNC #357: VCarve Pro post processor

nzlowie, 2016-06-10:
I'm having trouble finding a post processor in VCarve pro that bCNC can run. I create my drawings and g code in VCarve Pro but when I try to run the code generated I get error messages say that there are unrecognized lines or code. Can't remember the exact wording...

Any ideas please?

Thanks Dave

1bigpig, 2016-06-10:
+================================================
+
+ Grbl - Vectric machine output configuration file
+
+================================================
+
+ History
+
+ Who      When       What
+ ======== ========== ===========================
+ EdwardP  11/02/2015 Written from Grbl_mm.pp but
+                     set G20
+ EdwardP  11/02/2015 Commented out arcs as these
+                     slow GRBL performance appear
+                     interpolated anyway
+ BruceC   03/05/2015 Changed STOP command at end
+      of file. Readded ARCs
+================================================

POST_NAME = "Grbl0.9i (inch) (*.gcode)"

FILE_EXTENSION = "gcode"

UNITS = "INCHES"

+------------------------------------------------
+    Line terminating characters
+------------------------------------------------

LINE_ENDING = "[13][10]"

+------------------------------------------------
+    Block numbering
+------------------------------------------------

LINE_NUMBER_START     = 0
LINE_NUMBER_INCREMENT = 10
LINE_NUMBER_MAXIMUM = 999999

+================================================
+
+    Formating for variables
+
+================================================

VAR LINE_NUMBER = [N|A|N|1.0]
VAR SPINDLE_SPEED = [S|A|S|1.0]
VAR FEED_RATE = [F|C|F|1.1]
VAR X_POSITION = [X|C|X|1.4]
VAR Y_POSITION = [Y|C|Y|1.4]
VAR Z_POSITION = [Z|C|Z|1.4]
VAR ARC_CENTRE_I_INC_POSITION = [I|A|I|1.4]
VAR ARC_CENTRE_J_INC_POSITION = [J|A|J|1.4]
VAR X_HOME_POSITION = [XH|A|X|1.4]
VAR Y_HOME_POSITION = [YH|A|Y|1.4]
VAR Z_HOME_POSITION = [ZH|A|Z|1.4]

+================================================
+
+    Block definitions for toolpath output
+
+================================================

+---------------------------------------------------
+  Commands output at the start of the file
+---------------------------------------------------

begin HEADER

"G90"
"G17"
"G20"
"G0[ZH]"
"[S]M3"
"G0[XH][YH]"

+---------------------------------------------------
+  Commands output for rapid moves
+---------------------------------------------------

begin RAPID_MOVE

"G0[X][Y][Z]"

+---------------------------------------------------
+  Commands output for the first feed rate move
+---------------------------------------------------

begin FIRST_FEED_MOVE

"G1[X][Y][Z][F]"

+---------------------------------------------------
+  Commands output for feed rate moves
+---------------------------------------------------

begin FEED_MOVE

"G1[X][Y][Z]"

+---------------------------------------------------
+  Commands output for the first clockwise arc move
+---------------------------------------------------

begin FIRST_CW_ARC_MOVE

"G2[X][Y][I][J][F]"

+---------------------------------------------------
+  Commands output for clockwise arc  move
+---------------------------------------------------

begin CW_ARC_MOVE

"G2[X][Y][I][J]"

+---------------------------------------------------
+  Commands output for the first counterclockwise arc move
+---------------------------------------------------

begin FIRST_CCW_ARC_MOVE

"G3[X][Y][I][J][F]"

+---------------------------------------------------
+  Commands output for counterclockwise arc  move
+---------------------------------------------------

begin CCW_ARC_MOVE

"G3[X][Y][I][J]"

+---------------------------------------------------
+  Commands output at the end of the file
+---------------------------------------------------

begin FOOTER

"G0[ZH]"
"G0[XH][YH]"
"M5"

nzlowie, 2016-06-10:
Thanks for that. I see that this is for inch, is mm code any different?
Also having trouble running bCNC on my other laptop, installed Python but still won't load. Got it to run on my main PC ok...
Any tricks to this?

Thanks

vlachoudis, 2016-06-10:
@nzlowie I cannot help you much on VCarve. However inch or mm is not a problem in bCNC.
Your laptop problem, it is better to start bCNC from the terminal and see the messages that are displayed there. Maybe some package is missing

lalo-uy, 2016-06-10:
Try running it on the command line:

python bCNC.py

then you may see any missing lib

Guenni75, 2016-06-12:
I think Vcarve pro is from Vectric.
In Aspire 8 (from Vectric too), i use the GRBL8 post processor. It makes simple GCode.
Works perfect.

nzlowie, 2016-06-12:
Thanks for that, I use Vcarve 6.5 which is an old version and doesn't have the correct PP. 
The PP posed by 1bigpig works fine, I did change the G20 to 21 as I use mm.
Any chance you could post the Aspire version?

Thanks again guys.

Guenni75, 2016-06-23:
I said it before, the Grbl 8 PP works fine with Aspire.

vlachoudis, 2016-06-25:
I am closing the issue since it is not related with bCNC. maybe it would be nice to write some instructions on the wiki

closed 2016-06-25 by vlachoudis