vlachoudis/bCNC #358: Request - I want to add a physical button to the interface.

bobpombrio, 2016-06-13:
I have been asked to add a button that my user can press to activate the Pause function.  He wants it in case he suspect something is not quite right and he can inspect. Or because he see's a clamp is loose and he does not want to hunt for the Pause on the GUI.

I'm a developer but I do not know Python, yet give me a couple of days, so I am comfortable about making changes and testing before I submit the code to you for inclusion.  My thoughts were to use a simple push button that sets a signal high/low on a  RPI GPIO pin (lots left!) and have the GUI interpret it as a GUI Pause.

This request came about because I want the machine to have a E-Stop and he wanted to use it as a pause.  We all know where that conversation went and now I'm trying to add a Pause button. 

Thanks,
Bob P.

vlachoudis, 2016-06-13:
You have already this functionality on the GRBL side. Just connect a button to the PAUSE and RESUME pins of grbl.

bobpombrio, 2016-06-13:
I was worried that bCNC would not know about the grbl pause.

vlachoudis, 2016-06-13:
Not at all, they speak nicely the two. bCNC is sending the command to grbl and all actions are performed based on the status of grbl.

closed 2016-06-13 by vlachoudis