vlachoudis/bCNC #359: Question about camera usage

DaveBrauns, 2016-06-14:
I have a question about the probe camera alignment. In the wiki it is said to zoom in the image (section 2nd marker), but where? When I zoom in, only the workspace/coordinate system is zoomed, but I did not see any changes to the image. Did I miss any button/option?

vlachoudis, 2016-06-14:
Did you set the Camera Scale factor?
Depending on your camera and settings  you have to zoom a lot and then you will
see that the camera image will be zoomed as well. The zooming of the camera will try to match the zooming of the canvas using the Scale factor (how many pixels of the camera corresponds to 1mm in your canvas)

DaveBrauns, 2016-06-14:
Ok, I found the error. I only tried to zoom out and expected the image to shrink. Now I zoomed in, and after scrolling the mouse wheel 5 or 6 times the image enlarged. Seems the native cam resolution is higher as I expected.
Thanks a lot.