vlachoudis/bCNC #362: Autolevel and G02 - State error: Invalid gcode ID:36

sanotronics, 2016-06-18:
I will just link to the issue I opened in grbl's GH.

To make it short, I tried to mill a PCB with arcs in the corners and bCNC's autolevel feature changed the gcode G02 arc commands to G01 line commands and forgot to erase the R words, so GRBL throws the id 36 error, and the mill stops at the arc.

I ran the same program without the autolevel feature, and it went all the way through :)

[image]

vlachoudis, 2016-06-19:
Thanks for reporting. I will give a look

vlachoudis, 2016-06-19:
I forgot to remove the 'R' when the path was autoleveled, the IJK were removed :)
Now it should be ok. Could you test it?

sanotronics, 2016-06-19:
Yes, I'll try to test it later today.

Will I notice a difference in the smoothness of the arc, running it this way, which makes it into lines?

If so, then maybe I'll prefer running it without the autolevel feature, since I only care that it mills through the board up to the waste board, and for that I don't really care if the board is not leveled.

vlachoudis, 2016-06-19:
@sanotronics the accuracy of the arc->line conversion is controllable in the ```Tools->Machine->Plotting Arc accuracy``` by default is set to 0.01. This is means maximum offset between the line and arc of 10micrometer.

sanotronics, 2016-06-19:
Awesome :+1:

vlachoudis, 2016-06-23:
Great I am closing the issue then.

closed 2016-06-23 by vlachoudis