vlachoudis/bCNC #363: Small bug when mirroring g-code ?

lakeroe, 2016-06-18:
I think bCNC contains a small bug when mirroring g-code (horizontally or vertically).
Maybe this is related to bug #311.
If you can't reproduce the issue I can give more details.

vlachoudis, 2016-06-19:
Can you post an example

lakeroe, 2016-06-19:
Sure, an example is [Rohling.txt](https://github.com/vlachoudis/bCNC/files/322418/Rohling.txt) and also a screenshot before and after mirroring vertically.

[image]
[image]

vlachoudis, 2016-06-20:
At some point I was leaving as such the "M" commands in the processing and at some point I've added also the "G" which made this mess. Now it is fixed.

lakeroe, 2016-06-20:
Thanks a lot

closed 2016-06-20 by vlachoudis