vlachoudis/bCNC #364: Force Render out?

atrueresistance, 2016-06-18:
Is there a way to force rendering. I'm guessing it stops because it takes too long. In reality it doesn't take that long, but would really like to see actual margins. 

[image]
[image]
[image]

atrueresistance, 2016-06-18:
INF does not seem to do it. 

[image]

120 does 

[image]

vlachoudis, 2016-06-19:
Thanks, I've updated the infinite time (set internally to 5min :)

closed 2016-06-19 by vlachoudis