vlachoudis/bCNC #366: Strange behaviour withe eagle gerber files

jabss, 2016-06-20:
Hello,

First of all, congratulations for the great piece of work. Let me know if you need help translating to Portuguese.

I'm trying to create a simple PCB in my GRBL CNC with bCNC software running on a RPI3 (actuallly my first PCB using eagle PCB-GCODE plugin, so I might be doing something wrong) but I'm having a strange behaviour with the drilling file. The troubleshooting I was able to do, led me to think it may be related with bCNC and not so much with GRBL or PCB-GCODE.

So, I have two files, result of the PCB-GCODE outupt: The etching and the drilling.
I have successfully ran the etching, so I have all the tracks marked on the board.
Then, when I run the drilling the gandry starts moving in the right direction but when was about to stop and drill (first drill) it continues moving and it even goes out of the working area.
The strange thing is that bCNC "seems to be aware" of that because it shows the red circle correctly whenever it is (even when out of the expected working area). In other words: bCNC "knows" that it should work within a certain area, and its first move is to go completely out of it. (I don't know where it goes, because after a few cm out, it reaches the limit edge and then I have to do an emergency stop).

A detail that can make the difference to troubleshoot: In both files, after loading it, I do one CW rotation and then position the red circle accordingly. My feeling is that even if the picture in bCNC is correctly inverted, maybe the gcode is still not-rotated. Strangely, everything went well with the etching file.

I attach the files, just for reference.

Greetings from Portugal,
Joao Silva

[Drilling.tap.txt](https://github.com/vlachoudis/bCNC/files/323731/Drilling.tap.txt)
[Etching.tap.txt](https://github.com/vlachoudis/bCNC/files/323732/Etching.tap.txt)

vlachoudis, 2016-06-21:
I have the feeling that the "Tool change policy" is enabled but not configured.
Check the "Probe -> Tool -> Manual Tool Change -> Policy:"
if you don't want to change drills and automatically re calibrate them set it to "Ignore M6 commands"

jabss, 2016-06-21:
Hello,

I'm afraid the "Tool change policy" is disabled (with Ignore M6 commands option selected).
Anyway, I commented out ( ) all the M06 commands in the drill file and behaviour is the same.

Could this be a bug or some "problem between the chair and the keyboard"?

Thanks,
Joao

jabss, 2016-06-21:
Trying to follow up the code, I believe the gantry reaches the part

G82 X2.1 Y0.4 Z-0.032 F20 R0.1 P1

but when was supposed to stop and drill, it continues moving without changing direction. 

It never reaches the following command that is 

G82 X2.45 Y0.8

Strange, huh ?

vlachoudis, 2016-06-21:
Your gcode contains drilling cycles. G82 is not supported by grbl and it is expanded inside bCNC, maybe something is failing during the expansion. I will try to debug later once I have some free time. For the time being you can change the post-processor in pcb-gcode not to generate the drilling cycles.

BTW I cannot find the line you are mentioning.
The files that you uploaded contains the rotation or not?
Maybe the problem is in the rotation of the drilling cycles

You are certainly most welcome to do a Portuguese translation :)

jabss, 2016-06-21:
Hi,

The line is in the problematic (drilling) file, line 45.
I included both etching and drilling files just to show that the headers are very similar (maybe you were looking in the etching file). The etching file runs OK. I have at the moment the PCB etched by that file in my CNC and just need to run sucessfully the drilling part.

I'm afarid I'm not so sure what is rotation in the context of CNC, so I cannot tell for sure if the code has it...

Thanks

vlachoudis, 2016-06-23:
Ok. I've found the error and it is corrected now. Typical metric/imperial :) you were generating the gcode in inches, the gcode didn't have the G20 to switch to inches and the macro expansion of the canning cycles was not correcting for the inches scaling. Internally in bCNC all calculations are done in metric.
I didn't spot in the first place, since my machine was setup in metric and the gcode didn't have the conversion so in metric everything was working but scaled by 25.4. Now the new master has the correction.

vlachoudis, 2016-06-25:
I am closing the issue. If it not resolved with the last fix please reopen it

closed 2016-06-25 by vlachoudis

jabss, 2016-06-27:
Hello,

Confirmed that is solved.
Thank you,

Joao Silva