vlachoudis/bCNC #367: Network connection?

lmmeng, 2016-06-22:
Hi @vlachoudis,

How complicated will be to add a new "connection possibility" apart of the serial port (something like [websocket](https://pypi.python.org/pypi/websockets)) in the bCNC?
I'm thinking to try to use an [ESP8266](http://www.esp8266.com/wiki/doku.php?id=esp8266-module-family) to wireless connect the PC running the bCNC to the Arduino.
I was looking in the code, trying to implement it myself, but I don't have any chance to succeed (I've just started to read a Python tutorial :)).

Thanks,
Liviu

PS I've seen the other issue with Ethernet in name, but I'm not sure it is what I'm looking for.

Effer, 2016-06-23:
Hi @lmmeng , 
there are discussions about connecting bCNC to external world (software/hardware) in #155. 
But this should even be added in the bCNC wishlist at #310

bCNC is proven to be a good workbench to learn python anyway :)

lmmeng, 2016-06-23:
Hi @Effer,
thanks for pointing me to the right topic/issue. 
Looking through the #155, I found some interesting links to projects implementing similar (or better) ideas; I should study them further to see if it is what I'm looking for.
Meantime I'll continue to learn python. :) 

Regards,
Liviu

lmmeng, 2016-06-23:
Sorry, should learn to better use the git's issues system.

vlachoudis, 2016-06-23:
If I understand correctly you want to communicate with your machine via Wifi. If so, an interesting option (only on linux) is the socat command. You can make a network port appearing as a fake serial port, so you can communicate from bCNC.

lmmeng, 2016-06-23:
Hi @vlachoudis,
thank you for the suggestion. I've seen the socat option it in the #258 before opening the issue, but I was trying to avoid to many intermediate (extra) tools. 
Meantime, thanks to @Effer I've found other projects using socat and now I'm thinking to give it a try.
We can consider this issue closed for now.

Thank you both for your advices,
Liviu

closed 2016-06-23 by lmmeng

Bojesen2000, 2018-12-18:
I know it is old, but LaserGrcl got it running via esp8266.

Harvie, 2018-12-19:
We already have network transparency, see wiki for details:
https://github.com/vlachoudis/bCNC/wiki/Connecting