vlachoudis/bCNC #368: Progamm hangs

BlueOrangeLive, 2016-06-30:
Hello vlachoudis
I have downloaded and installed just bCNC aktuel down. The program does not start by, hangs and must be stopped only with xkill. I have on my second computer the same problem. At the second start it ending itself. After a computer reboot again the same problem.
Here is a picture of it.
[image]

Greetings Jürgen

vlachoudis, 2016-07-01:
Please verify you are using python2 and not python3 and check the console for any message indicating missing packages

BlueOrangeLive, 2016-07-01:
Hello. 
bCNC version before a par days works for me. Only now the current version does not.
Python 2.7 is installed. I use Fedora24.
Once Downloaded today. Again the program depends.
This time starts from the console, the error message here .:

[xxxxxx@xxxxx2016-f24 .bCNCprg]$ ./bCNC
Traceback (most recent call last):
  File "./bCNC.py", line 2441, in <module>
    application = Application(tk)
  File "./bCNC.py", line 429, in __init__
    bFileDialog.append2History(os.path.dirname(Utils.getRecent(i)))
  File "/usr/lib64/python2.7/posixpath.py", line 122, in dirname
    i = p.rfind('/') + 1
AttributeError: 'NoneType' object has no attribute 'rfind'

BlueOrangeLive, 2016-07-01:
When I delete the lines 429 +430  in bCNC.py, then starts the program.

428		# Filedialog Load history
429		for i in range(Utils._maxRecent):
430			bFileDialog.append2History(os.path.dirname(Utils.getRecent(i)))

vlachoudis, 2016-07-01:
@BlueOrangeLive many thanks. It was a recent addition to display the last directories visited in the dialog frame. I forgot to put a check if they were less than the maximum.
Now it is corrected.

closed 2016-07-01 by vlachoudis