vlachoudis/bCNC #369: Workspace change without pressing Buttons

Guenni75, 2016-07-01:
I am working with G54 and G55.
If you are working with G55, press STOP and then START again, the Workspace changes to G54 without changing the State of the Button G55 to G54.
If you press a Control Button to move the Gantry the State of the Button changes.

Same thing after loading a new GCode file while G55 is active. The G55 changes back to G54 but the the Buttons didn't change. 

I ruined some pieces because the Workspace was wrong and after moving the Gantry i wondered why the Workspace changed.

vlachoudis, 2016-07-01:
Is it the latest version you are using?
Normally when you stop a job, bCNC is forced to do a software reset to clear any pending command in the grbl buffer. After the reset grbl is switching to g54. bCNC is aware of that only after issuing a $G command which it does periodically.
Some time ago I've added a saving and restoring the G state of the machine before and after the Stop, but yesterday I've pushed a lot of changes so maybe I screw it up.

Guenni75, 2016-07-04:
I loaded the last version and it is still the same.
If you are on G55 and you load a new file, it changes to G54 without changing the Button state.
If you press one of the arrows to move the gantry, the G54 Button changes to pressed.

vlachoudis, 2016-07-04:
@Guenni75 I am bit puzzled since loading a file doesn't send any commands to grbl, so there should be no change at all.

When you send commands to grbl (by pressing the arrows or from the command line) periodically when grbl is idle and **not running** anything bCNC will issue a $G command to read back the state and refresh the button status.

During running, bCNC doesn't send any $G command to inspect the state, so if the WCS was changed there or just before the run, there is no way to know the state.

Guenni75, 2016-07-04:
I did it in this way:

bCNC Start.
Pressing Connect (GRBL Idle)
Homming (G54 pressed)
Pressing G55
Loading a file (G55 is still pressed)
Press Start
Press Stop (in this moment the Wpos numbers change to the G54 values, but the G55 button is still active)
Pressing Arrow right (more then ones) and the Button changed to G54.

If i press the Stop button at active G55 Button after starting bCNC changes Wpos to the G54 values and back to G55. That' OK.

The Problem appears not all the time, only when i load large files (8000 lines for example) the Workspace change after pressing STOP doesn't work right. At small files it's ok.

vlachoudis, 2016-07-04:
Can you post the output of the terminal window to see what happens.
I've tried exactly what you are reporting and I cannot reproduce it.

Guenni75, 2016-07-05:
For example this:

ok
G55
$G
ok
[G0 G55 G17 G21 G90 G94 M0 M5 M9 T0 F0.]
ok
[G0 G55 G17 G21 G90 G94 M0 M5 M9 T0 F0.]
ok
G90
ok
T1
G17
G21
G0Z20
G0X0Y0S12000M3
G0X37.966Y-3Z6
G1Z-0.692F90
G1X0F180
G1X-0.1381Y-2.9968
ok
G1X-0.2784Y-2.9871
ok
G1X-0.4205Y-2.9704
ok
G1X-0.5639Y-2.9465
ok
ok
Grbl 0.9j ['$' for help]
G91G0X10
G90
ok
ok
[G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 F0.]
ok

After STOP and pressing one Arrow, the state changes to G54.

vlachoudis, 2016-07-05:
@Guenni75 can you ensure you have the latest version of bCNC (download the master) not the released one. As I can see the "ok" are not in sync with the commands, while on the latest version they should appear in sync.

Guenni75, 2016-07-05:
I downloaded the master.zip this morning.

vlachoudis, 2016-07-10:
Can you confirm that the large file doesn''t have a workspace change command inside?

Guenni75, 2016-07-30:
I checked some of the files and there is no workspace change.

Guenni75, 2016-10-10:
I made a makro on a button with a workspace change and the Workspace-Buttons still don't change.

vlachoudis, 2016-10-10:
@Guenni75 the Workspace buttons are updated only when the grbl is interrogated with a $G command. So you should add a $G in the end of the macro.

closed 2016-10-10 by vlachoudis