vlachoudis/bCNC #370: bCNC ignores some PCB tracks

jabss, 2016-07-06:
Hello,

I'm using eagle and PCB-GCODE to produce GCODE files to ran in bCNC.
I realized recently that if the circuit is more-or-less complex, some tracks are missed in bCNC.
This gets obvious if we define ground-planes in Eagle, but even avoiding it, in some cases some tracks seem to be missing. I wasn't sure if the problem was in the PCB-GCODE or in bCNC, but I loaded the GCODE into the CAMotics program and the missing tracks are there, so I'm assuming that the process, up to the GCODE creation is OK.
At some stage I thought that bCNC was somehow defining the ground-planes on its own, but in the example below, there are two distinct tracks that are merged...

Example:
This is the PCG-GCODE graphical output: http://pasteboard.co/8h6hE7zvd.jpg
This is what appears in bCNC: http://pasteboard.co/8hbOteCbs.jpg
And the result is: http://pasteboard.co/8hi8AKNps.jpg
The GCODE: https://www.zerobin.net/?9266f9109b856cf7#K1/+tqtt8KBHtKmdzQlGSNJQUALIrf3QOAE1GQyhipQ=

Is there a limitation in the size of the GCODE that bCNC can handle?

By the way, not sure if relevant, I'm runing bCNC in a RaspberryPi3.

Any idea?

Thanks,
Joao

vlachoudis, 2016-07-06:
It is only on the drawing because of the rendering time cut set. You have 2s set on the top right corner of the canvas. Change it to a bigger value and it should show correctly.
bCNC has no limitation of the size of the file, but for big files on slow machines the rendering takes a lot of time so it stops after a certain interval.

vlachoudis, 2016-07-08:
@jabss could you please confirm if this was the problem and close the issue?

jabss, 2016-07-08:
Hi,

I initially though the problem was more than just the image display because
the missing tracks are also missing in the physical PCB (identified in the
photo). However, I ran everything again with 30s rendering timeout and it
worked flawlessly.

So I guess we can close this case.

Thanks,
Jabss

vlachoudis, 2016-07-08:
Sorry I didn't look all your images. It seems it is not a rendering problem, but for some reason the file was not loaded entirely. When I loaded in bCNC you gcode it displays everything correctly on my laptop even with the smallest rendering time cut.
But as I said the rendering cut is only for the display and it wont affect what is sent to grbl.

closed 2016-07-08 by vlachoudis