vlachoudis/bCNC #371: Translation to Portuguese

moacirbmn, 2016-07-07:
I did translation to Portuguese, and is almost complete, how do uploud? which folders should I send?

Effer, 2016-07-07:
Hi @moacirbmn , 
with git you can open a pull request and it will contain all needs.

If you have follow this How To: https://github.com/vlachoudis/bCNC/wiki/How%20to%20contribute%20a%20translation 
you should have end now with this structure in the bCNC folder: **\bCNC\locale\pt\LC_MESSAGES**
(I'm not sure if pt is the correct language code for Portuguese) and inside you should have 2 files **bCNC.mo** and **bCNC.po**
This is all it needs I believe.

moacirbmn, 2016-07-07:
I'm putting more translations but can not upload the folder to the github (the translation I'm doing is for Brazil's Portuguese)

vlachoudis, 2016-07-08:
@moacirbmn if you don't succeed in making a pull request, post your files on this issue and I could upload them.
BTW what is the difference of Brazilian Portuguese and the ones from Portugal?

moacirbmn, 2016-07-08:
@vlachoudis It appears to me it "Uploads are disabled.
File uploads require push access to this repository. "

I will try for send you the file already ready to put in the repository.

to answer your question, the difference between Portuguese of Portugal and Brazil are few, usually in few technical terms and adjectives.

moacirbmn, 2016-07-08:
@vlachoudis I am still new at development with github, How can I send the file to you?

vlachoudis, 2016-07-10:
How to create a pull request: https://help.github.com/articles/creating-a-pull-request/
The files you could upload them to any file sharing site, or even on google drive and share the link

vlachoudis, 2016-07-10:
It is strongly preferable to make a pull request it will help you for subsequent changes.

moacirbmn, 2016-07-10:
@vlachoudis sorry, I'm very inexperienced and I think I messed up when I made the request, so I decided to put in my google drive the folder with the translation.

https://drive.google.com/folderview?id=0ByoRhH0i6j6sUWpkNlRQZ1BxZW8&usp=sharing

vlachoudis, 2016-07-11:
@moacirbmn thanks. I've just added it

Effer, 2016-07-11:
Thanks @moacirbmn. 
To keep it updated, learn some git commands will help a lot. There are even some GUI that make it easy.
Anyway if you have doubt feel free to ask (even PM) or post somewhere the updates files.

moacirbmn, 2016-08-13:
They can make an update of the translation in Portuguese?
I made over 25% of translation

https://drive.google.com/folderview?id=0ByoRhH0i6j6sUWpkNlRQZ1BxZW8&usp=sharing

Effer, 2016-08-15:
Merged. Thanks @moacirbmn

closed 2017-05-11 by vlachoudis