vlachoudis/bCNC #372: Problem don't start

el3kim, 2016-07-07:
elk@elk-hp:~/bCNC-master$ ./bCNC

Traceback (most recent call last):
  File "./bCNC.py", line 2443, in <module>
    application = Application(tk)
  File "./bCNC.py", line 193, in __init__
    page = cls(self.ribbon, self)
  File "/home/elk/bCNC-master/CNCRibbon.py", line 106, in __init__
    self.register()
  File "/home/elk/bCNC-master/ControlPage.py", line 1102, in register
    (DROFrame, ControlFrame, StateFrame))
  File "/home/elk/bCNC-master/CNCRibbon.py", line 125, in _register
    w = f(self.master._pageFrame, self.app)
  File "/home/elk/bCNC-master/ControlPage.py", line 142, in __init__
    DROFrame.dro_mpos   = Utils.getFont("dro.mpos",   DROFrame.dro_mpos)
  File "/home/elk/bCNC-master/Utils.py", line 260, in getFont
    font = makeFont(name, value)
  File "/home/elk/bCNC-master/Utils.py", line 212, in makeFont
    font.configure(family=value[0], size=value[1])
IndexError: tuple index out of range

Effer, 2016-07-08:
Hi @el3kim ,

can you post here your **.bCNC** User File? You can reach it from here:
[image]

Or in your user folder as described here: https://github.com/vlachoudis/bCNC/wiki/Configuration

I suppose some font settings is wrong. If so we'll add some security check.

vlachoudis, 2016-07-08:
@el3kim It seems that the size was not supplied in one font and I didn't have a protection on that. I've just added one, so now you should be able to start and correct the problematic font

el3kim, 2016-07-08:
Thanks you.
Great work.

closed 2016-07-09 by vlachoudis