vlachoudis/bCNC #373: GRBL reported error unsupported command

burtbick, 2016-07-10:
Hi,  I have a gcode file that works fine with ChiliPeppr but it is a fairly large file, 13000 lines.

The only issue with ChilliPeppr was that I forgot to turn off the 3D simulator / viewer so it was pretty slow as far as UI response was concerned.

So I thought I would run the file in bCNC.  I had v 0.9.2 on my CNC computer and when I started the run I had some initial movement and then it stopped with GRBL saying unsupported command.  It didn't make sense based on the error message the command was G21G90G64G40  which was the first few command from line 4 in the file: G21 G90 G64 G40 

And like I said the gantry had moved before it choked.  There is a T4 M6 command and I thought that I had commented it out before I started the run just in case.

I exited bCNC and started ChiliPeppr up again and the file ran to completion just fine, no problems.  ChiliPeppr pops up that there is a tool change command found any you have to hit pause to restart after changing the tool (if needed).  

Is there something more that needs to be done when processing files with tool change commands in them?

I looked Probe->Tool and under the Manual Tool Change the Policy is to ignore M6 commands.  does that make sure that no M6 command in a gcode file is sent to the controller (GRBL in this case)?

If it isn't the M6 command that was causing GRBL indigestion then there must be something else.  Either another command that ChiliPeppr is filtering out, or some hiccup that sent GRBL something that it didn't recognize when running the job with bCNC.

I've downloaded the latest bCNC and I can try to repeat the test tomorrow using that version, but wanted to double check to see if there was anything else that needs to be done to prevent sending M6 commands to GRBL?

Here's the start of the file in question, the first 17 lines.

( Made using CamBam - http://www.cambam.co.uk )
( RoundoverTest 7/9/2016 5:44:10 PM )
( T4 : 3.16 )
G21 G90 G64 G40
G0 Z3.0
( T4 : 3.16 )
T4 M6
( 3DSurface1 )
G17
M3 S1000
G0 X31.58 Y16.31296
G0 Z1.0
G1 F50.0 Z-0.2
G1 F360.0 Y0.0
G2 X30.0 Y-1.58 I-1.58 J0.0
G1 X3.67123
G0 Z3.0

====  When you say that bCNC ignores M6 commands does that mean that it will not prompt you to change a tool if an M6 command is found in the gcode?  Or does it just mean that if an M6 command is found in the file that it will not pass it on to GRBL?

Thanks,

burtbick, 2016-07-10:
I did a bit more digging.  It appears that it is probably the G64 in the line 4.  What is interesting is two things:

1 - GRBL did some moves of the gantry before choking.
2 - ChilliPeppr masks this by apparently filtering out the commands that GRBL doesn't support, which can be a bad thing if it is a command that would be critical to the machining operation.

G40 is also not supported by older versions of GRBL but was supported as of v 0.9i and I am at 0.9j on my controller.

But I still want to make sure that I understand how the new tool change support works.  If I have not calibrated and configured (since I haven't yet set up any kind of detection mechanism) what happens when an M6 command is found in the gcode?  Will bCNC prompt, or indicate, the tool change request?  Or will it just keep trucking and skip sending the M6 to GRBL?  I hope that if the automatic processing for tool change isn't set up that bCNC will at least prompt you to give you an opportunity to change the tool.
It would be nice, but of course harder to keep in sync with different versions of GRBL, if bCNC could warn about unsupported commands in the gcode.

vlachoudis, 2016-07-10:
In the old versions of bCNC when you have an error from grbl you can click on the status message and it displays the problematic line. In the new master, the terminal commands are now in sync with the replay from grbl so you can immediately spot the problematic command.

**What is important to remember is the bCNC when grbl returns an error it will stop.**

However, you can configure bCNC, in the .bCNC section [Error] you can set how bCNC to act in case a problematic command is found. e.g. if you set G64=1 it will skip all G64 lines.

Concerning the tool change policies you can give a look on the wiki.
Briefly
- **Send M6 commands** will send the M6 commands to the controller as is in the gcode
- ** Ignore M6 commands** it will strip out the M6 command from the gcode
- **Manual Tool Change (WCS)** it will stop to the Change location. Wait for you to change tool do a probing cycle, and continue. Optionally you can pause it after the probing to start up manually the spindle if needed. WCS = it will update the WCS z coordinate after the probing to match the new tool
- **... (TLO)** as the (WCS) but it will set the Tool offset instead
- **... (NoProbe)** it will simply pause to change the tool and no probing will be perfomed

burtbick, 2016-07-10:
Ah.  I didn't spot the NoProbe option when I was looking at it late last night.  In this case with the G64 issue as it was included on a line with other critical commands the correct thing is to stop as it is doing.  It was just that it was different from how ChiliPeppr does it, but in reality it would be better if ChiliPeppr also stopped or told you that there was an issue with the gcode if it is going to skip or filter the bad command out.

In any case I located a post processor for GRBL on CamBam so I didn't have to create my own, and that works and plays fine with GRBL now as it doesn't output the G64 command.  Note that it does output G40 but GRBL .9i and above supports that now.

Thanks,...

closed 2016-07-10 by burtbick

vlachoudis, 2016-07-10:
@burtbick  I agree, I find it dangerous to continue when there is an error