vlachoudis/bCNC #375: Execute on Job Complete & ALARM

DasWookie, 2016-07-15:
This is a feature request to be able to execute a command on event.

It would be terrific if there was a means of executing a system command (or script) on job complete and ALARM. Even better would be if there was the option to execute different commands on success, fail, alarm, feed hold, door, reset, run, etc. This way, when a job is completed or needed attention, one could be notified by the system (for example by text and/or email) that the job has finished it's run and/or alarmed out or whatever, and the unit needs some attention.

I run my xCarve in a lights out configuration for most of it's work, but it's machining aluminum so these are LONG run times. I try and remember to set a timer to remind myself to go check on it, but invariably I forget, and end up letting long times go by where the machine ends up sitting idle. If I had a means of sending myself a notice somehow via a system command or scripe when events occurred, then I could more easily manage production.

Effer, 2016-07-15:
I proposed something similar some times ago and then python script inside gcode  was introduced. It's still missing the ability to run the code only when is running on machine (not while is edited). 
Anyway the event driven system will be a nice addition.

DasWookie, 2016-08-05:
Python script INSIDE gcode is well beyond what I was hoping for. Mostly just the ability to "on event" trigger fire off a shell script or bat file. I run lights out, so some form of alarm function would allow me to easily integrate with an alert script which would be able to page me... or control a light tower... or send me an email... or or or... I tried to take a look at the code to see where I might integrate something like this, but the ramp up on getting familiar with the codebase proved to be more time consuming than I had available at the time.

vlachoudis, 2016-08-06:
I've added a new [Events] section in the ini file. For the moment it contains two event types
`onstart = `
`onstop = `
which are system commands to be executed before the start of a run and after the end
e.g.
`onstart = date`
`onstop = python /home/user/alarm.py`

More over I added the long standing request of @Effer to have a variable to be True during running and ignored during editing. It is called `running` and if you add python commands like
`%if running: print "This printout will be executed during run"`
`%print "This is executed any time the gcode is parsed"`

Could you please check if they work correctly. And if yes please document them on the wiki

Effer, 2016-08-06:
Finished some early test and it's a great addittion IMO.
There's some problem if the evaluated line is wrong. It raise strange beahviour. I've made a PR #400  that try to fix it.

DasWookie, 2016-08-06:
Been experimenting with this, and I'm having difficulty getting the os.system call to work at all. I can run my test alarm script from command line, and call them from a test prog, but for some reason it's just not working from bCNC. 

In my bCNC.ini I have the values set for:
>pi@bCNC:/inbox/Projects/CNC/Software/bCNC/bCNC $ grep onst bCNC.ini
>onstart = ./alarm.py
>onstop  = ./alarm.py

I have tried variants like:
python alart.py
python ./alarm.py
/usr/local/bin/python /home/user/alarm.py
et cetera...
all to no avail. The test script is never executing via the os.system calls on start or stop.

My test script I'm using to troubleshoot is:
>#!/usr/bin/python
>import os
>onStart = "./alarm.py"
>if onStart:
>       	try:
>               	retval = os.system(onStart)
>       	except:
>              		pass
>print retval`

So I'm at a loss as to what's going wrong here. My python version is 2.7.9.

My test alarm.py script is attached here:
[alarm.py.txt](https://github.com/vlachoudis/bCNC/files/405119/alarm.py.txt)

Effer, 2016-08-06:
@DasWookie maybe you are editing the wrong settings file?  
Is not the one in the bCNC directory but is one file in user directory named **.bCNC**

DasWookie, 2016-08-07:
Thought I responded earlier, but don't see it here. yes, that was the problem I was editing the bCNC.ini is the program dir, not my user directory level .bCNC file. The relative pathing with the file was a little confusing as well... with ./Events/alarm.py being local to the program dir, not the user dir, took me a little bit to figure out as well... but I got that sorted.

The "running" variable confuses me a bit... am I to understand that maintains scope across the call, such that I could use it in the event script? Or is that just in scope in bCNC itself? Sorry, not new to programming but am a total newb with python... but I can't see how that could share scope across the os.system call.

I'm loving the end of run report as well. That's got just the information I was after!