vlachoudis/bCNC #376: Math on DRO for WPos axis entry

DasWookie, 2016-07-15:
One of the really nice features of many professional DRO heads is the ability to perform simple and quick calculations on them. The biggest one I find I'm using on the Mill and Lathe ALL the time are inputting something like  a "\ 2 {enter}" to center once I've found the edge (or edges) of the work piece or a hole. Other times I find I end up pulling out a calculator to compute my offsets, which can invariably result in a human error when I then enter the result into the WPos for that axis. Being able to do something like a ( or ) ^ which would correctly follow / enforce PEMDAS would be terrific, but even just having the ability to do multiplication, division, addition, and subtraaction would save me from most of the times I'm having to do calculations outside of bCNC to enter into a WPOS axis.

Effer, 2016-07-15:
Wow fresh air! @DasWookie this is really a nice idea.

DasWookie, 2016-07-15:
I don't know about that... I just keep trying to modify the WPos like I do on the mill and lathe, and then realize or have to remember... I can't. :(

vlachoudis, 2016-07-17:
You can already do some calculations in the command line and set the offsets. For example
```set [2+3] [2*sind(5)] 0```

vlachoudis, 2016-08-05:
The patch of @Effer was modified to accept also the variables of the program. For example
if you enter in X: `xmin` it will set as x the lower x margin of the gcode

closed 2016-08-05 by vlachoudis

Effer, 2016-08-05:
@vlachoudis more than a patch, mine was something that need to be patched! :)
Now it works as it should. Very well done.

DasWookie, 2016-08-05:
I sync'd to latest to test it out... but I'm getting an error on launch:
$ ./bCNC
Traceback (most recent call last):
  File "./bCNC.py", line 2374, in <module>
    if sys.version_info.major != 2:
AttributeError: 'tuple' object has no attribute 'major'

My python version is 2.6.6:
$ which python
/usr/bin/python
$ /usr/bin/python --version
Python 2.6.6

I can launch the prior version I was running no problem...

vlachoudis, 2016-08-06:
@DasWookie could you upgrade your python to the latest v2.7