vlachoudis/bCNC #380: Program freeze

juanquy, 2016-07-21:
On  Xubuntu 16.04 program freeze when I try to open a .dxf file.
Any ideas?

regards
JC

Effer, 2016-07-21:
Hi @juanquy ,
can you post the dxf file?  Just drag and drop it here. 
Any message from the prompt?

juanquy, 2016-07-21:
I will do it tonight . But before I have a question. My CNC have to be
online before I can I open and manipulate any file? Because when it happen
my machine was offline.

Regards and thank you for ur help
Juan C

juanquy, 2016-07-22:
Hi: Im trying to attach the .dxf but I got this error message "Unfortunately, we don’t support that file type."
I will zip it and try again. 
No messages from the console, the program just freeze.

juanquy, 2016-07-22:
Hope you can open it.
regards 
JC
[dxfile.zip](https://github.com/vlachoudis/bCNC/files/377382/dxfile.zip)

Effer, 2016-07-22:
Thanks @juanquy , I'll have some test later. Anyway you can open and edit file even if bCNC is offline.

Effer, 2016-07-22:
At least in windows it work. It takes some time (a couple of minutes) but it looks nice:

[image]

If you are using a slow hardware maybe it take a longer time.  Can you try to let it load for some minutes?

juanquy, 2016-07-22:
I will try later maybe it's because my hardware is not so fast . Thank u

juanquy, 2016-07-23:
Same, I got the program running on a PPC G5 Quad with Ubuntu 12.04 64bit and same happen, I will try to open a different file (stl or gcode) to see.
I have Inkscape installed on this Computer (PPC) and I can open/manipulate dxf files. So I dont know what is going on here.
regards
JC

vlachoudis, 2016-07-23:
The problem is that bcnc is using an adaptive splitting of the splines to line segments, and it takes to long time for your dxf. I will try to optimize it and simplify the algorithm

juanquy, 2016-07-25:
However I think something is wrong here, because the program freeze opening the file. Maybe it is arch I'm using it. 
Kernel Linux 4.4.0-31 PowerPC-smp ppc64 bit
Its a multiple processor kernel. 
The computer is a PowerMac Quad G5 (PPC) 2.5Ghz , RAM is 2G.
This is enough hardware to handle CNC operations.
I'll keep researching  about this on my free time.
regards and thank you for your work for the CNC community 
JC

Effer, 2016-07-25:
I case you need or just for test here the file saved as ngc by bCNC
[Lancette.zip](https://github.com/vlachoudis/bCNC/files/380822/Lancette.zip)

juanquy, 2016-07-26:
@Effer  the file open perfectly, one more question, how I can declare/change the size of my work area?

Effer, 2016-07-26:
@juanquy 
Tool->Machine->Travel x/y/z

; maximum travel of your machine in the three axes
These values will define the WorkArea in the main canvas (enabled/disable from button)

[image]

**Be aware: for historical reasons values are always referenced from top/right corner even if your home switch position is in the bottom/left corner.**

Also the View will be correctly displayed only after homing procedure complete.
This values will not be in sync with grbl settings so you need to adjust even that for your machine works correctly.

juanquy, 2016-07-28:
[image]

Now I will send you this file and I wanna you guys tell me why I can open this dxf but not the one I send you before. The software just open it in just a second.
Here is the dxf in a zip 
[Gear Sample.dxf.zip](https://github.com/vlachoudis/bCNC/files/387817/Gear.Sample.dxf.zip)

I have to rename Sample.DXF to Sample.dxf because when I tried to open it with the extension in capital , bCNC don't see it as dxf, actually you cant find it in the open file window. After I rename it it was ok you can see the file and open it.
Another question, I was trying to do some live work but the first thing that happen to me was that I have $H disable because I do not have Home and limit switches in my Shapeoko 2, I have it but I haven't time to mount and wiring it. So I usually set (reset Zero)
X=0,Y=0 and X=0 on the left bottom corner of my work area and I been doing this without problems.
So its a way that I can set my Zero coordinates on the bottom left corner with bCNC and a command to Return to my Zero ?

vlachoudis, 2016-08-02:
It seams that there is some numerical precision error appearing only on the Mac version for the specific file. It would be nice to debug, but since I don't have a Mac I will send you a modified version to dump a lot of debugging messages to pinpoint the problem

juanquy, 2016-08-02:
Oki can do that.

juanquy, 2016-08-02:
Do you need my email to send me the file?

vlachoudis, 2016-08-03:
Yes please. If you want you send me a private email on my address in the about box.

juanquy, 2016-08-04:
Its ok my email is juanquy@gmail.com

jgregorc, 2016-09-07:
Hi,
i also have a "program freeze" problem, but it seems to have different origin. I would appreciate any comments, suggestions or ideas.
I have the following hardware: Home made CNC with grbl running on grbl shield and Chinese clone arduino UNO. The spindle is 1.5kW three phase with matching inverter. All cables are shielded, with shield connected to the ground. I run bcnc on 32bit winXP python 2.7. The problem is that program stops "freezes" in the middle of the run. I can not pinpoint why, but it seems to be related to the spindle (if i only run gcode without turning the spindle on the problem does not appear). Please see error.png as it shows the error in cmd and error.txt for more details. I don't understand what it says.
If someone can comment on this issue i would really appreciate it.

Regards,

JG

[error.txt](https://github.com/vlachoudis/bCNC/files/459994/error.txt)

[image]

Effer, 2016-09-07:
Hi @jgregorc ,

I've encounter that exception some times ago after hitting stop to my machine, never during a run. The relevant part is the **writeTimeoutException** that seems to point to some communication error between bCNC and grbl.
I have some benefits, maybe 6 month ago, with a commit that improve the serial handshake, but mainly after upgrading my pyserial installation. Different from your config, I'm using win7 that is supported from the most recent pyserial version, for WinXp latest version is number 2.7, see [here](https://github.com/pyserial/pyserial/issues/132). You can check your pyserial version from python:

```python
import serial
print serial.VERSION
```

That said the problem can be related to some electric noise from your spindle inverter.
* If all cable are all shielded (also usb) try to connect only one side to ground and let the other side not connected to anything. 
* If you give command to spindle drive with arduino I/O try to remove all the connection and give start and reference with external command. Later you could use an relay to electric separate grbl from inverter.
* You can try to slow down speed communication (both grbl and bCNC).
* Change arduino clone with one with better serial IC

Let us know.

jgregorc, 2016-09-08:
Hi @Effer 
thank you for your suggestions. I will try them and report the results.

Regards,
JG

robcop2000, 2016-09-10:
I have the same problem

onekk, 2016-09-10:
Maybe it is related to some bug in the pyserial package that print out some garbage at the reconecction, i have upgraded the pyserial supplied with the os with the latest 3.1... and it solved most of the connection problem.

jgregorc, 2016-09-11:
OK. So i have tested the first option as suggested by @Effer.

- I connected only one side of spindle cable shield to ground. I have been running cnc (different jobs) for about 5 hours with only one "freeze". So there is some improvement.
- I also ordered new original arduino uno.
- the pyserial upgrade will have to wait for a while (i need to buy new computer)

Anyway i still think that the problem is with electric noise coming from the spindle and the low voltage communication on USB. Perhaps i should change the UNO+gshield to something else. Any ideas (for hobby machine)
Regards, JG

onekk, 2016-09-11:
I'm an user of an eShapeoko, and I've had some problem in the early running of the machine, so I've rebuild all the electrical part, using:

Shielded cable for the motors, try to connect the shield only at one end and leave the other end unconnected (the motor end has to be left unconnected according to some literature)
Shielded cables for the limit switch and all the wirings, like the buttons for Grbl
Reworked all the ground cables and reunited them in two connections on the metallic chassis enclosing the eletctronic
Powering the Arduino with a 12V power supply and not relying on the 5V USB, and using some ferrite beads and also two capacito one electrolictic and one ceramic of 100pf  near the arduino and the stepper shield power input (both cina made).

Using some capacitor near the limit switch inputs to avoid the noise.

Now I'm using my CNC without connection problem during working, and I use a Kress monophase spindle of 1050W of power and a shopvac of 900W attached to the same power outlet of the controller as both the spindle and the shopvac are controlled by two SSR driven by the CNC Shield (a cinese copy of the protoneer).

genrally power inverters are well shielded, but also the cables that go from the inverter to the motor have to be shielded as the noise generated from the inverter can be quite high, check the routing of the cables and avoid long parallel running of the power cables of the spindle.

the Noise is a beast, some literature say to use metallic (ferrous) box to shield also the magnetic component of the noise, aluminium enclosing shield only the electric component. 

Hope this it will help with your machine.

Regards

Guenni75, 2016-09-12:
Since a week maybee (2 or 3 BCNC Versions back) i have the freeze Problem.
Sometimes it happens at the start of the job,  sometimes in the middle.
If it happens, i have the Write Timeout in the Python window.

Guenni75, 2016-09-12:
He did it again...
[image]

Guenni75, 2016-09-12:
And again...
[image]

robcop2000, 2016-09-12:
Same line error I have in my machine

Guenni75, 2016-09-12:
I am trying an older BCNC Version (April).
Sometimes the router stops a second, but he finishes the job.

Effer, 2016-09-12:
@Guenni75 thanks to point it out.

robcop2000, 2016-09-14:
@onekk  can you post some picture of the electrical soloution of your machine.
Thanks

onekk, 2016-09-14:
Yes i could post some photos, but it is not too difficult,

put a 0.1 uF or 0.33 uF capacitor between ground and every limit switch pin near the input on the controller board.

This will generally iimprove the noise rejection of the limit switch input, maybe some other solution it's better, but this is very simple and can be done soldering one end of the three capacitor togheter and the other end can be soldered to the plug terminal that go in the borad or inserted in the screw terminal.

This simple solution have served me well, I have done a quick an dirty solution soldering a piece of bread board holding the capacitors and the pin strip acting like a sort of connector between my board and the machine wiring. (ver 0.9 of Grbl change the pin of Z limit switch if i remember well)

For the other  electrical solution, i have used some barrel connector with screw terminal and punt in the two teminal the capacitors, (one 100 uF electrolitic, one 0.1 uF polyester and one 100 pf ceramic, following a technique used in the ham radio ham) and the cables are passed trough a ferrite beads.

For the ground connection, simply I have left the cable as short as possible and groupend togheter in two single pack attached with screw and groower waser to the metallic chassis (in my case and old and very heavy 1990 era PC case).

As soon I can I make some photos of the arrangements, in few days I have to rework some connections in the case.

Regards

onekk, 2016-09-19:
Here the photo, sorry for the quality but my phone is not the best.

[image]

[image]

[image]

closed 2017-01-02 by vlachoudis

miomaox, 2017-01-10:
Bcnc funziona correttamente per AUTOLEVELLING ; SCAN DEL PROBE ; SPOSTAMENTI DEL CONTROLLER ; praticamente tutto OK, ma quando vado a incidere il contour di un PCB, lo inizia correttamente ma verso la metà del lavoro si inceppa e non risponde più a nessun comando ad eccezione dello spegnimento. A volte funziona lo stop, allora porto a zero nuovamente il mandrino e inizio una nuova incisione del contour, tuttavia, anche in questo caso il lavoro non viene mai portato a termine.
Per il GCODE io uso questo programma online: http://rapid-pcb.com/
Ovviamente ho bisogno di aiuto!
Grazie a chiunque possa prendere atto di questa mia richiesta.
1000 Cordialità
miomaox