vlachoudis/bCNC #381: Z motor vibrates when I run a homing

thomasfourmond, 2016-07-21:
Hello !

I'm trying to use bCNC with GRBL (pibot : http://www.pibot.com/connect-cnc-hardware-for-board-rev2-x) . When i run homing or when i use Z control, nema 23 motor vibrates (only Z).
But when i open STL file (from EASEL - Inventable software for example) and run, Z axes is not vibrates.  
 
Did you have some ideas to help me ? I really want to use bCNC !

Bye ! And many thank for this incredible software !

Effer, 2016-07-22:
Hi @thsfo ,
can you post the **terminal** output when the Z vibrates? (is the right most tab in bCNC)
Maybe the homing speed or jog speed is set too high for Z axis. But it should't be anything related to bCNC. Do you try to run some simple job with bCNC?

vlachoudis, 2016-08-05:
I am closing this issue since it seems unrelated to bCNC

closed 2016-08-05 by vlachoudis