vlachoudis/bCNC #382: order.optimize doesn't account for toolchanges

RicII, 2016-07-22:
The ordering in the editor, especially optimize is not aware off tool changes between blocks which
makes the clone & tile operations less useful.

Ideally it should group blocks on tool to minimize tool changes and optimize the resulting path accounting for the tool change (& optional probe) position. (using the manual tool change macro)

vlachoudis, 2016-07-29:
Optimize simply moves blocks of code to minimize the g0 patps. All commands inside the block will remain as is therefore to changes etc will not be rearranged. Maybe in the future can be mage not smart. So for the moment it is reconnected to apply the optimization selectively

vlachoudis, 2016-08-02:
Just for clarification. The optimization was done for optimizing the imported dxf before generating the gcode. Optimizing/reordering  already existing gcode can lead to unpredicted behavior, it has to be done with care.

closed 2016-08-02 by vlachoudis

RicII, 2016-08-04:
Noted, I got the impression that the tool change support was a later addition as well. The cleanest way to implement this would probably to create an other hierarchy level in the editor.