vlachoudis/bCNC #384: Small Bug

FelipeArturoLopez, 2016-07-27:
While I was testing the software I came aware that it was not possible to change the Y axis of the workspace. Everytime that I change the coordinate and press enter nothing happened. I looked the code and in ControlPage.py, in function setY there's a problem because it sends a gCode that is not recognizable. Even after running the scipt it showed several times that the problem was that instruccion. I commented the instructions and leave only the instruction: self._wcsSet(None,self.ywork.get(),None) and it worked. I hope it helped a little bit. Even I checked the file in the repository to see if there wasn't the bug but I found out it still there.

Effer, 2016-07-27:
Thanks @FelipeArturoLopez ,
I've spotted the same and reported in PR #379 . I'll be soon fixed.

Effer, 2016-08-01:
It is corrected from the latest pull request

closed 2016-08-02 by vlachoudis