vlachoudis/bCNC #385: start job from gantry position

466, 2016-07-28:
Hi there, 
when i start a job, the gantry moves back to zero and goes all the way from there to the gcode's location.
Which means, it does not move above several obstacles, but crashes right into them ... .
this already broke 2 endmills and 2 pcbs .... .

What do i have to do in order to get bCNC start the job right from where the gantry is right now?
I am on the up to date github bCNC version btw.

Thank in advance!

vlachoudis, 2016-07-29:
Most probably your gcode contains g28 or g30 instructions "go to origin" but you have not set them. Could you check your gcode?

466, 2016-07-29:
Ok, thanks for your reply.
I do not have G28 or G30 within my code, but i do have P1:

Recently i added limit switches, which do work for homing, but due to some bad wiring, i decided to turn them off,  before i start a milling job (via $21=0 and $22=0)
After homing, the gantry is at X-0.99 Y-0.99 Z-0.99

After disabling homing and limit swichtes, i use the probe feature to set a new Z zero, which is right above the surface.

My gcode does contain multiple P1 commands, which moves the gantry to X-, Y- & Z- zero.
I think i should set P1/G54 Z- zero way above the workplate (where it was after homing), but how do i use the probe and auto level feature here?

vlachoudis, 2016-08-02:
What do you mean by p1 moving the gantry? P1 is for pause 1 sec

Guenni75, 2016-08-08:
If you want to start from the gantry position, you have to set x and y to 0 before you start.
GCode: G92 x0 y0  ; for example or press the X=0 Y=0 Buttons.
I don't understand the Problem.

closed 2017-01-02 by vlachoudis