vlachoudis/bCNC #386: keypad jogging

poesel, 2016-07-29:
First: thanks for writing bCNC - great program!

I've bought a wireless key pad for jogging. Works great. Two issues:

1. how do I redefine the keys? I would like to map 8,2 to x and 4,6 to y. Currently its the other way round which is a bit inconvenient because my setup is perpendicular to that. Also can't just turn the keypad because then up/down would then be left/right. Also 7 is already labelled 'home' so I'd rather use that than 1.
2. it would be nice if the keyboard buffer would be deleted after a numpad key has been released. Because queued keys lead to crashes. :(

Thanks!

vlachoudis, 2016-08-02:
1. This is a nice proposal, I will try to implement it.
2. From Python/tkinter I don't have access to manage the keyboard buffer. Maybe you should lower the repeat and delay of your keyboard settings

poesel, 2016-08-02:
1. cool! :)
2. I think you can. Most concise answer I got from here [https://rosettacode.org/wiki/Keyboard_input/Flush_the_keyboard_buffer#Python]:

def flush_input():

    try:

        import msvcrt

        while msvcrt.kbhit():

            msvcrt.getch()

    except ImportError:

        import sys, termios

        termios.tcflush(sys.stdin, termios.TCIOFLUSH)`

vlachoudis, 2016-08-03:
2. This works only on MS windows for a terminal application. bCNC is a tk application, where all the input events keyboard and mouse are handled by tk.

poesel, 2016-08-03:
The msvcrt part is for Windows, termios for Unixes. Although I don't know if that has any relevance together with tk.
There is a solution for tcl on the same page [rosettacode.org](https://rosettacode.org/wiki/Keyboard_input/Flush_the_keyboard_buffer#Tcl). Again, don't know if that helps you or not.
If not then I'll have to wait until grbl gets proper jogging support. Thanks for the efforts!

vlachoudis, 2016-08-03:
I would prefer also to wait for grbl, rather making a temporary solution and then to be forced to re-write it.

vlachoudis, 2016-08-05:
@poesel in the new master I've added a variable in the ini called "swap" in the [Control] section.
Default is 0: Left/Right on X, Up/Down on Y
Other options
1: Left/Right on Y, Up/Down on X (CW rotation of 90deg)
-1: like 1 but direction inverted (CCW rotation of 90deg)
I don't know if it needs more choices 
You have to set it in your ${HOME}/.bCNC  file and not in the bCNC.ini.

poesel, 2016-08-07:
Works as advertised - thanks a lot!

Would it be possible to configure the home button, too? Or to disable it altogether? Because it is a function I never use and hitting it accidentally may cause a crash.

poesel, 2016-08-07:
Sorry, found a bug.
I added "swap = -1" under [Control] in my .bCNC. This worked the first time but after restarting the [Control] section is empty except for the swap entry. I cannot move the gantry with the keypad anymore although the keypad works. Same with my backup copy.
I remember there was at least one other entry but I can' remember which. I guess the bCNC.ini file is the template for the one in home and it has several entries. Which of these is necessary?

poesel, 2016-08-10:
Things are working as normal again. I've tried to reproduce the bug but couldn't. So please disregard the bug report.

vlachoudis, 2016-08-10:
@poesel could you please document on the wiki and I am closing the issue

closed 2016-08-10 by vlachoudis

poesel, 2016-08-10:
Yes, will do. But what do you mean with ‚Canvas‘?

https://github.com/vlachoudis/bCNC/wiki/Jogging

poesel, 2016-08-11:
done

vlachoudis, 2016-08-11:
@poesel thank you. Canvas I mean the drawing window on the right with the gcode and the workspace