vlachoudis/bCNC #391: bCNC never get to 100% done

fantasiiio, 2016-08-02:
Hi there,

I started using bCNC I like it !
But I have a problem. Whatever the job I start, it never finish to 100%. I mean, the job itself if finished, but the progress bar is not 100% and I cannot do anything because all buttons are disabled. And I'm stuck there. I can reboot the grbl controller but it's like a workaround.

Here I did an autoLevel, the job finished correctly and it returned to home. But as you can see, the progress bar is not to 100%. It does the same thing with any gcode. I tried with grbl controller and there no problem.
[image]

Can you help me ?

Effer, 2016-08-02:
Hi @fantasiiio 
can you post here a zip with the gcode?
The ouput from the terminal would help a lot. 
If you hit the stop button something happens?
And last but not last, do you modify the footer code (Tools->Machine->Footer code)?

Hope this will help to find what's wrong.

fantasiiio, 2016-08-02:
Hi, thanks for the answer.

For the gcode, I tried as simple as:
G20
G90
G94
F3.00
G00 Z0.1000
G01 X0.5Y0.5
G00 X0.0000Y0.0000

And I got the same result:
[image]

For terminal I have a "Idle,WPos:-0.001,-0.002,2.538,Buf:0,RX:0,Lim:000" five time a second so, I cannot really see what's happening in the terminal. Is there a way to hide these "idle" messages ?

If I hit the stop button, all controls become enabled again, but the cnc doesn't respond anymore. I have to reset it.

I don't think I modified the footer. Do I need to ? Here a screenshot of my machine parameters:
[image]

Thanks

vlachoudis, 2016-08-03:
Did you modify the status reporting string on '?' from grbl?

fantasiiio, 2016-08-03:
No. I did not write the '<' because the html parser interpret it as a tag...

it's more like :
< Idle,WPos:-0.001,-0.002,2.538,Buf:0,RX:0,Lim:000 >
without space beside the right angle brackets

But, it seem that grbl still have something in it's buffer. It report it's state but does't accept any commands. It doesn't respond to basic commands like '$$'

Do you think it's possible ?

fantasiiio, 2016-08-03:
Also, why do you think the progress bar writes: 9 [9] but only 80% ... 9 of 9 is 100% ? isn't
And there is a 33%.. what is that percent mean ?

vlachoudis, 2016-08-03:
Your status reply is missing the MPos:xxx,yyy,zzz information

fantasiiio, 2016-08-03:
I see. What is the difference between wpos and mpos?  Is bcnc is waiting for mpos ?

fantasiiio, 2016-08-03:
Ok I got it !
It was indeed missing mpos. Maybe I changed the 'status report mask' by mistake. I setted the mpos bit flag to 1 and... It worked !

Also, in bcnc, the status was always 'Not connected' and I never seen the gantry position before.

Thanks you very much for you support !

vlachoudis, 2016-08-04:
Thanks. Anyhow we have to change the logic and to make it more robust on status report change.

closed 2016-08-04 by vlachoudis