vlachoudis/bCNC #402: manual spindle control ?

fantasiiio, 2016-08-06:
Hi,

In the control section, there is a spindle on/off button and a slide bar for the speed. 
It does nothing, no commands are sent to grbl. Is it normal ? Is there something I have to do before ?

Effer, 2016-08-06:
Hi @fantasiiio ,
I believe there's a problem here:
Spindle and other commands are locked, untill the state from grbl has been updated from bCNC. 
If you connect the serial port automaticaly on startup it seem the state is not reported untill a switch to workspage (g54,g55 ..) is send.
If the grbl state is missing you should see the control tab with empty fields (distance,tool,feed etc).
When the state has been reported the filed will be filled with some data and Spindle commands will be enabled.

fantasiiio, 2016-08-06:
Hi, even if the auto connect at startup is turned off, all grbl states are missing. When I click on a workspage, states appears but I still cannot control the spindle. 

All jogs commands works well.

Effer, 2016-08-06:
Which grbl version are you running?

fantasiiio, 2016-08-06:
The lastest: 0.9

Effer, 2016-08-06:
That's strange, I was getting the same problem as you, but after the state has been updated, everything run as it should. I can see the M3/M5 sent in the console. Tomorrow I'll have more test.

levis0, 2016-09-26:
I found this problem (does it is problem?) too - after warm startup of bcnc spindle did not works. but after pressing unlock button it works without any other manipulations. this restriction is in bcnc code, from terminal M3 Sxxx works ok.

vlachoudis, 2016-09-26:
The commands should be send when the machine reports Idle as state. Unless if there is a bug and before any command bCNC doesn't update the state.

closed 2017-01-02 by vlachoudis

phil123456, 2017-12-23:
just had the issue, clicked on reset seemed to solve the problem
now instead of connected it says idle and I can turn on the spindle

FelixWeichselgartner, 2022-01-08:
For me I had to send another Z+ or similar command so the spindle started