vlachoudis/bCNC #403: Z shifting after 1000s of moves

auhopu, 2016-08-11:
I posted the same on GRBL's issues page, but I am copying here as well as it might be relevant to bCNC:

I tried cutting 4 simple circular pockets and used jscut's ramping feature (i.e. instead of straight plunging, it ramps into the required Z - or creates helical movements in my case of circular pockets, to optimize tool life). The outcome is ~9k lines of gcode, half of which involve Z moves.

After running the gcode, I noticed that it went back to the given safe height of 3mm (Z WPos showing indeed 3), but it was actually about twice higher. Here's my train of thought to debug this.

- Wrong $102: This was my initial thought. However, a wrong $102 would only cause problems of scaling (i.e. if it was higher/lower, my Z moves would compress/expand proportionately). However, moving from Za to Zb and back to Za should go to the same Za, even if this is out of scale. In my case, Za shifts over time! Then I tried to manually give 30~40 Z moves and they were all consistent (and, btw, correct in scale too). Z0 would always go back and kiss the waistboard. So whatever happens, happens after 1000s of Z moves.

- Probing issue: I am probing in different locations on the waistboard and when going G1 Z0 it is spot on. No probing/zeroing issues.

- Skipped steps: This is the next area to question. However, no sign/sound of skipped steps. Furthermore, what makes me exclude this one is consistency. It would always shift the same. Even without load when "air cutting".

- Waistboard not level: Turns out also irrelevant. Even though I do plan to face mill the waistboard soon, I do my Z shifting comparison on the same X,Y (before running the gcode and in the end).

- Tool slipping: That would be a possible explanation. If for some reason the tool did not engage in the collet tight enough, and the cut was too aggressive, then theoretically the tool could slip upwards and present exactly what I experience (this is a good visualization to imagine what is actually happening). However, this is out of question too... as, again, upwards Z shifting eventually occurs (by the same amount) when "air cutting" !

I am left with two areas (that I can think of) as possible culprits.

- Buffering: If for some reason some gcode was taken out of stream, I would see Z shifts. However, the fact that I am cutting downwards and the shifting happens upwards, makes this less likely. Also, I am using bCNC which is supposed to be the (or at least one of the) best performers in streaming massive gcode.
EDIT: Hm... If downward Z moves were dropped at buffering, could bCNC think the spindle was lower than it was so that subsequent upward Z moves would make it shift further up? It is the consistency of this happening over and over in the same pattern that puzzles me.

- Rounding: This is the only one I have not debugged. I am using an ACME and therefore my $102 is ugly:
(200step/rev * 2microstep/step) / (25.4mm/in / 12rev/in) = 188.976microstep/mm

Which makes me wonder if 1000s of Z roundings are enough to cause Z shifting. One thing I thought to try today is to throw my $102 out of scale with a nice round number (say 200microstep/mm) and see if the same gcode still causes Z shifting.

Btw, I am using the X-Controller with GRBL version 1.0c.20151109 .

Any ideas, recommendations are very welcome.

vlachoudis, 2016-08-11:
I've noticed while typing you are using 2 microsteps/step. I had a similar problem in the beginning in X/Y directions and I when I increased it to 8 or 16 the problem disappeared.

What you report is that possibly grbl is losing the position, because it reports that it is at +3mm while in reality is higher. My understanding is that if grbl is properly set for the machine, it should not lose position no matter whatever g0,g1,g2,g3 command you give, even if you skip lines etc. So even if there were errors in probing, buffering etc you might have some unwanted steps, but the final location of the gantry should correct.

I believe that either you are losing steps (wrong settings in grbl or hardware problem) the air-cut problem would exclude that you are engaging aggressively the tool into the material, or tool is worn out.
Or maybe as you are pointing some numerical precision problem.

Do you mind sharing your gcode. I could test air-cutting on my machine to see If I can reproduce it.

$122 z acceleration maybe incorrect?

Effer, 2016-08-11:
In my case was solved lowering z acceleration.

auhopu, 2016-08-11:
As Sonny pointed out in [this](https://github.com/grbl/grbl/issues/1054) thread, this seems to be cause by helical plunging and high acceleration. I will need to further experiment with my $11, $120, $121, $122

My defaults were:
$11=0.02
$120=500
$121=500
$122=50

and by looking at GRBL's [defaults.h](https://github.com/grbl/grbl/blob/edge/grbl/defaults.h)
`#define DEFAULT_X_ACCELERATION (500.0*60*60) // 25*60*60 mm/min^2 = 25 mm/sec^2`
`#define DEFAULT_Y_ACCELERATION (500.0*60*60) // 25*60*60 mm/min^2 = 25 mm/sec^2`
`#define DEFAULT_Z_ACCELERATION (50.0*60*60) // 25*60*60 mm/min^2 = 25 mm/sec^2`

I'm confused as to whether they should be 500-500-50 or 25-25-25 as suggested by the comments...

@vlachoudis 
gcode file attached. 
[1_waistboard_8.5mm_insert_13-14mm_deep.gcode.zip](https://github.com/vlachoudis/bCNC/files/413734/1_waistboard_8.5mm_insert_13-14mm_deep.gcode.zip)

@Effer 
Do you remember your before/after values?

auhopu, 2016-08-12:
I performed an experiment last night and indeed it's the helical plunging that causes this and indeed it is affected by the acceleration values. Surprisingly, low acceleration performed worse. More details in [this thread](https://discuss.inventables.com/t/z-shifting-after-1000s-of-moves-on-x-controller/26739/4).

Thanks everyone for sharing your thoughts.

closed 2016-08-12 by auhopu

Effer, 2016-08-14:
@auhopu I was unable to access my cnc till now.
If you still need it, here my grbl settings:
$110=2500.000 (x max rate, mm/min)
$111=2500.000 (y max rate, mm/min)
$112=1500.000 (z max rate, mm/min)
$120=250.000 (x accel, mm/sec^2)
$121=250.000 (y accel, mm/sec^2)
$122=100.000 (z accel, mm/sec^2)

Previously $122 was 250 as I can remember, but too high for my custom mechanics and motor torque.