vlachoudis/bCNC #404: <Suggestion> loadable buttons

onekk, 2016-08-14:
Working with my CNC machine and bCNC i have increased te number of buttons to 21 and set some useful function on them like M8 and M9 on some "buttons".

my thinking is:  

- some settings are helpful on some conditions, as in my case when i have to test the bed for flatness an test the position of some holes in the bed (with the purpouse of testing if the bed is incorrectly placed).
 
in everyday job these setting are not useful, so if you can save/load the button settings, it will be a very nice feature to use.

the most simple way would be to use an external file to hold the button definition and hold it "on demand" maybe via a list in the "control interface" i think tha it wold be simple to add a single option to .bCNC.ini as 

Buttons=internal / external

and if it is external load the file the default place and scan maybe a hidden directory (.bCNC or maybe .config/bCNC) fot the relevant files maybe with .btn extensions and present them in the interface as a dropdown list ready to be changed "on the fly".

But I didn't know ho the internals of bCNC permit this in some "not too difficult" manner.

TIA and Best Regards

Carlo D.

vlachoudis, 2016-08-14:
You can start the program with -i file.ini to create multiple ini files. So you could store your special configuration there?

onekk, 2016-08-16:
This can be a viable solutions, but for me is "second best".

I was thinking with your proposed solution, that it wold be nice to have a sort of reload bCNC.ini mechanims, without losing the machine positions MPOS and WPOS.

But in this case you have to implement a form of selection of ini file in the interface. 

Most of my GCODE related to the button are positioning at relevant places in the workspace but maybe with different G54...9 coordinates space, even with some G53 for Machine position, for some job a "set" of coordinates and button assigned.

In my opinion the best will be:

1) adding two buttons for M8 and M9 in the main control window, near the spindle controls. (if it will not make the controls too crowded)

2) make some sort of "selectable sets" of buttons.

Maybe asap i will try to modify some code and submit to you.

Regards

Carlo