vlachoudis/bCNC #406: bCNC autoprobe feature

atrueresistance, 2016-08-16:
I am not sure if bCNC has this feature or not.

https://youtu.be/bhZdbgM6S70

I know you can probe the different directions, but not sure about angle correction. I know I would find this super helpful. So does bCNC have this feature?

Effer, 2016-08-16:
Not exactly the same, the [Camera Alignment](https://github.com/vlachoudis/bCNC/wiki/Probe-Camera-Alignment) allow to adjust a job to fit some reference points. And so it corrects not only the angle, but also the scale of gcode. What is missing is a the z reference. For that you can adjust it manually or using the probe/autolevel feauture.
Anyway it's a nice feature and maybe it can be added as another way for probing.
Thanks for sharing.

vlachoudis, 2016-08-18:
You can already do it as a macro in one of the programmable buttons

atrueresistance, 2016-09-02:
Yes, I have the macro buttons configured for zeroing my different sized bits. More or less I was suggesting angle correction feature for a future release.

vlachoudis, 2016-09-03:
Angle correction , you have already in the Orient frame under probe. Not in such a way but it could be done. All the machinery is there